Ultimaker/Cura #15388: Supports touching build plate starting on layer #2

Pro4791, 2023-04-29:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Creality Ender-3 V2

### Reproduction steps

Turn off build plate adhesion with supports enabled.

### Actual results

No brim, skirt, or raft present on the model or supports causing them to start printing on the second layer and not adhering to the build plate.
[image]
[image]

### Expected results

No brim, skirt, or raft present on the model with supports having a base on the first layer. Should be like the photos below but with no skirt around the model, only the base on the supports.
[image]
[image]
Works as intended in Cura 4.13.1
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Support CURA.zip](https://github.com/Ultimaker/Cura/files/11359704/Support.CURA.zip)

GregValiant, 2023-04-29:
Thanks for the report.
This is duplicate and a lot of people are noticing this bug.  The workaround is to enable either the skirt or brim.  When Bed Adhesion is set to "None" then the support brim gets disabled and the support doesn't show up until the second layer.

Asterchades, 2023-04-30:
I think I may have some insight into this one.

[image]

This is the first layer of supports on my test model using Brim adhesion, so everything is working as expected. Note specifically where the Support is being printed as regular Support, rather than as the Support Brim.

[image]

This is the same model, same settings, but now with None adhesion. The Brim part of the Supports is gone, but the regular Support part is still present.

[image]

All I have done here is disable the Support Brim setting. It would appear that the presence of _any_ Build Plate Adhesion is being adjusted by the more global setting at the detriment of the Support-specific setting, which is arguably a regression on previous functionality (that grid really doesn't seem like it's going to stay put).

GregValiant, 2024-06-22:
@Pro4791  I believe this problem has been addressed in 5.7 versions.  Can this be closed now?

closed 2024-07-07 by github-actions[bot]