Ultimaker/Cura #15389: Cura 5.3.1 - Creality Ender 3 S1 Pro Nozzle Size Support

Apie56, 2023-04-29:
### Is your feature request related to a problem?

Upon selecting the Creality Ender 3 S1 **Pro** printer profile, there is no option for nozzle sizes other than 0.40mm,

### Describe the solution you'd like

Like the Creality Ender 3 S1 printer profile, nozzle size options range from 0.20mm to 1.00mm

### Describe alternatives you've considered

The only way around this is to use the Creality Ender 3 S1 printer profile for my Creality Ender 3 S1 **Pro**.

### Affected users and/or printers

It will affect users with the Creality Ender 3 S1 **Pro**.

### Additional information & file uploads

[image]

GregValiant, 2023-04-30:
The definitions come from the community and whoever submitted that definition forgot to include the nozzle files.

In the Cura installation directory find "...\cura\resources\variants\creality".  Scroll down and find the "creality_ender3s1pro_0.4.inst.cfg" file.  Open it in a text editor.  It's only a few lines long and you need to change two of them to your new nozzle size and then do a "Save-As" using the exact same naming format but with your new nozzle size.

Here I have change what was "0.4" to "0.6" and the file name would be "creality_ender3s1pro_0.6.inst.cfg"
[general]
definition = creality_ender3s1pro
name = **0.6mm Nozzle**
version = 4

[metadata]
hardware_type = nozzle
setting_version = 21
type = variant

[values]
machine_nozzle_size = **0.6**

Repeat that as needed to get your nozzle size options for your printer.

Apie56, 2023-04-30:
Awesome, thank you, that fixed it.

closed 2023-04-30 by Apie56