Ultimaker/Cura #15390: Skewed box

mmoeller55, 2023-04-29:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

Have a basic box that I need to enlarge along the X axis. 

### Actual results

Box can be enlarged, but the resulting print is skewed (not rectangular)

### Expected results

Rectangular box

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[9x3x2 smooth top.zip](https://github.com/Ultimaker/Cura/files/11360282/9x3x2.smooth.top.zip)

GregValiant, 2023-04-30:
Thanks for the report.
It looks like you will need to rotate the model so it fits your build plate.  The rotation must the the last transformation you make.
Cura works on the "bounding box" of a model and the bounding box doesn't rotate with the model.  Instead, the bounding box changes size to fit a rotated model.  The XY axes of the model stay square to the bounding box which always stays square to the world.  That means that when you stretch a rotated model you are stretching the bounding box (which is square to the world) and so it skews what is inside the box because the model is no longer square to the world.
I stretched first and rotated 2nd and the model is square cornered.
[image]

Here I rotated first and then stretched it and it skewed.
[image]

So the rule is "rotation last".  The only alternative would be to have the part coordinate system rotate with the part.  All three axes would need to be considered and things get very complicated very quickly.  Rotate last.

I'll leave this open and as a bug report for the Cura team but my thought is that for the forseeable future this will remain "expected behavior".

mmoeller55, 2023-05-01:
Thank you for correcting my error. I have reprinted the item, using your
suggestions, and it has perfectly square corners.

GregValiant, 2023-05-01:
There ya go.  I know it can be a head scratcher, but it isn't like a CAD program where you can align the coordinate system any way you please.
I'll go ahead and close this then.  Let me know if you think that is premature.

closed 2023-05-01 by GregValiant