Ultimaker/Cura #15391: Cura 5.3.1 Slicing Failure on Scaled Up Chest

enewbold, 2023-04-30:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

FLSUN V400

### Reproduction steps

Loaded Basic-Chest-Optimised-V1.stl and selected generic PLA material. 
Scaled model up to 400% .
Selected Skirt and Supports On Buildplate .
Attempted slicing and it failed. 
[image]

### Actual results

Nothing. Slicing failed.

### Expected results

Slicing should have worked.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11360480/cura.log)
[FV_Basic-Chest-Optimised-V1.zip](https://github.com/Ultimaker/Cura/files/11360481/FV_Basic-Chest-Optimised-V1.zip)
[image]

GregValiant, 2023-04-30:
Thanks for the report and the files.
Here is a report on the model from [formware.co](https://formware.co/OnlineStlRepair)
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 2 Non-manifold edges (?)
--> 4 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 4 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 46625ms ------
-> Vertex count changed from 164711 to 165334 (+623)
-> Triangle count changed from 329418 to 331044 (+1626)

Even with the repaired file, slicing took a long time so I cancelled it.  In order to get this to slice I had to:

- Repair the model.
- Move it away from the mid-point of the build plate in both the X and Y.
- Rotate it 90°.
- Change the "Initial Layer Height" to 0.31

After all that if finally sliced.  I'd call that a bug.  I'll mark this as a duplicate and a slicing error.
[image]

enewbold, 2023-04-30:
Thank you for the update.  I think I'll probably not print this model.
(smile...)

Ed Newbold

GregValiant, 2023-04-30:
I'll leave this open as it's a good example file for the Cura team to use as a fly swatter.

closed 2024-04-03 by HellAholic