Ultimaker/Cura #15392: Not change print speed or flow with extention "modify Gcode - "ChangeAtZ 5.3.0"

MarcelloMazza, 2023-04-30:
### Application Version

5.3.0

### Platform

Window 10

### Printer

FLSUN V400

### Reproduction steps

I tried to use the G-code modifier to vary the speed and the print speed with the "changeAtZ" function at a specific layer but it doesn't work, as well as the flow change doesn't work. None of the variations are highlighted in the preview by color of speed or flow. Due to the speed variation, the printing time does not change either. Obviously I had checked "enabled"

### Actual results

None of the requested variations are triggered in the G-code

### Expected results

None of the requested variations are triggered in the G-code

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Not change for print speed at layer over  5mm.zip](https://github.com/Ultimaker/Cura/files/11360919/Not.change.for.print.speed.at.layer.over.5mm.zip)

MarcelloMazza, 2023-04-30:
I checked the gcode and the requested change is made. I don't know why this variation is not represented with colors and is not implemented in the printing times but the important thing is that it is carried out in the printing.

GregValiant, 2023-04-30:
Thanks for the report.
Post processors are fairly simple and do make changes to the Gcode but they generally have no function to change the listed "Print Time Estimate".
Post Processors by definition run after the model has been sliced and after the Cura Engine has put together the gcode.  When you select the "Save" button the raw gcode is passed to the post processor, the changes are made, the altered gcode is passed back to Cura, and then the gcode is immediately written to the gcode file.  There is no function to update the preview to the "Post Processed" gcode file.  If a model is on the build plate then the Preview is based on the Slicer settings.

If you were to open a Post Processed Gcode file in Cura - the Cura Preview would reflect the changes that a Post Processor made PROVIDED the changes involve commands that the Preview portion of Cura understands.  
This gcode file was created with two instances of Change At Z running.  The speed was change to 150% at layer 25 and then back to 100% at layer 90.  You can see that the changes show up in the Gcode preview where they did not in the regular Slicer preview.
[image]

Here is a preview of an antenna in Cura and ArcWelder is enabled.  This is the raw slice data and is based on the Cura settings and G0/G1 moves.
[image]

Here is the Gcode file opened in Cura.  Since ArcWelder runs as a post processor, and since the Preview capability of Cura does not understand the G2/G3 moves added by ArcWelder - the Preview looks a LOT different than the slicer preview did.
[image]

This is all "expected behavior" and is currently the way it works.  I'll remove the bug label but leave this open so you can respond here.  A feature request for a button to show the post-processed gcode preview might be an option.

MarcelloMazza, 2023-04-30:
Perfect; many thanks for your explanation.

GregValiant, 2023-04-30:
I'll go ahead and close this then.

closed 2023-04-30 by GregValiant