Ultimaker/Cura #15398: First layer infill speed

Sledgar, 2023-05-01:
### Is your feature request related to a problem?

I have to lower the first layer speed because especially the edges of the outline dont stick but the "infill" of the first layer prints with the same slow speed despite having no need for it.

### Describe the solution you'd like

An option to set outline and infill speed seperately for the first layer

### Describe alternatives you've considered

changing the gcode or sitting next to the printer amd setting the print speed to 200% while its printing the infill.

### Affected users and/or printers

General problem

GregValiant, 2023-05-01:
When you say "Infill" do you mean Infill or are you actually talking about the Bottom Skin?  It's a very rare thing to print "Infill" on the first layer.

You mention a problem with "not sticking" on the first layer and that is usually it's own problem.
One thing you can do now that may help is to increase the "Initial Layer Flow".  Try it at 110% so start.  That will give you more squish between the bed and the nozzle and you may get better adhesion as a result.

Sledgar, 2023-05-01:
Yea i didnt really meant infill. My working was kind of weak here. I meant the zick zack lines to fill in the outline which gets layed down first. I noticed that I can print about twice the speed on the filling than on the outline. I increased flow and line width already on the first layer and generally have okay adhesion albeit not perfect but on bigger prints the first layer costs me just so much time because I only print at 20mm/s whereas the filling could be easily printed at double that speed or even higher

GregValiant, 2023-05-02:
There are speed settings for top/bottom and for Initial Layer and for Walls as well as Number of Slower Layers.  Between them you should be able to get a combination that will work.  The alternative is to customize.
You will have to make manual changes to the gcode.
Open the gcode in a text editor.
Search for "TYPE:SKIN" and then right below the existing line...
;TYPE:SKIN
M220 S150
That will change all speeds to 150% but skins go down last on any layer.  Your 20mm/sec will go to 30mm/sec.
Next, search for LAYER:1 and do the same sort of thing...
;LAYER:1
M220 S100
That will drop your feedrate back down to 100%.

The number one thing you should do is work on the bed adhesion.  I put my initial layers down at "Print Speed" which can be from 75 to 125 for PLA on my stock Ender.  A clean build plate and I have no problem with stiction.

5axes, 2023-06-01:
As Workaround I'm using a Postprocessing script : https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/FastFirstInfill.py