Ultimaker/Cura #15404: Non-Ultimaker Printers Missing

inspectionsbybob, 2023-05-01:
### Application Version

5.3.1.

### Platform

Win11

### Printer

Non Ultimaker

### Reproduction steps

Installed the latest release
Tried to ADD a Printer
Non Ultimaker
Non Networked

                Nothing shows except the Ender-3

NOTHING

### Actual results

No List of printers

### Expected results

Expected to see a LIST of Creality printers, and the option to add one manually.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No Files to load as I can't load a printer

GregValiant, 2023-05-01:
If you can start Cura then got into the MarketPlace and see if you have "StartUp Optimizer" loaded.  If you do then uninstall it.  It keeps additional printers from showing up in order to get Cura to load faster.
If it is not installed then we'll need a Cura.log file to trouble shoot.  Use the "Help | Show Configuration Folder" command to go to the location of the log file.  Zip it and post the zip folder here.

inspectionsbybob, 2023-05-02:
I re-loaded clean, no add-ins...
Nothing showed.

Loaded 4.13.x and everything was there...

I will double check that the startup optimiser wasn't left over from an
earlier install but I don't think so...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016