Ultimaker/Cura #15406: (5.3.1) Merged models not slicing as one merged mesh

truegret, 2023-05-01:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Creality 10S

### Reproduction steps

1. Import two models ( 2 halves of a large case/box )
2. Oriented and aligned the two halves, butted end-to-end
3. Selected both halves, right-click, and "Merge"
4. Sliced combined or merged mesh

### Actual results

1. Slicer sliced the models as individual models in situ

### Expected results

1. After slicing, seeing continuous shells all around the single merged mesh model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

It is worthy of noting that while I prefer to use Cura, I tested the same with a competitor slicer, and successfully accomplished the desired result.
[20230501 Merged Models Dont Slice as One.zip](https://github.com/Ultimaker/Cura/files/11367622/20230501.Merged.Models.Dont.Slice.as.One.zip)

Thank you ....

GregValiant, 2023-05-01:
Thanks for the report and the files.
There have been bug reports regarding the Mesh Fixes tool "Remove Mesh Overlaps".  Maybe it is involving other facets of Cura.
I used MS 3D Builder to merge your two halves into a single model.  That app also has a decent repair utility and it noticed some errors within the model.
No reason to wait for a fix for this (or you can use your PrusaSlicer gcode).  I'll mark this as a duplicate and the Cura team will take a look.
[Export_Merged-HousingA-B.zip](https://github.com/Ultimaker/Cura/files/11368240/Export_Merged-HousingA-B.zip)

truegret, 2023-05-01:
Thanks, Greg.

I have used 3D-Builder as well to "repair" models and it does a very good job.  I have also used netfabb, that also works well.  Since you mentioned 3D-Builder, the Prusa slicer seems to remove the mesh overlaps as well.  In general, I tend to have 3 slicers (Simplify3D, Cura, and Prusa) I try to keep more or less similar profiles where each slicer has something unique or a slightly "better" feature to use.  Simplify3D is used much less now that they begged for $50 toward their last long overdue upgrade.

For the record, Ultimaker Cura has been my No 1 since your team has always been very responsive, produce a better product, and strive for continuous improvement.  Thanks!!