Ultimaker/Cura #15407: (5.3.1) Need ability to manually enter a precise angle while using the rotate tool

truegret, 2023-05-01:
### Is your feature request related to a problem?

It is not a problem, but rather a feature that would aid in precision placement or rotation of a model

### Describe the solution you'd like

As a hobbyist and 3D modeler I would like to precisely rotate a part into position by manually entering an angle.  The current method limits this by using a left-click and mouse drag with Snap Rotation unchecked.  It is desired to select a rotation axis and enter a desired angle.

### Describe alternatives you've considered

As a hobbyist and 3D modeler, and engineer, using the mouse with left-button depressed, it is too difficult to get the desired angle.

### Affected users and/or printers

It would be beneficial to all that desire to precisely rotate a model.  It would also be beneficial to have multiple models that need to be merged, so allowing a precise, manually entered angle via the keyboard is desirable.

### Additional information & file uploads

Thank you for the consideration

GregValiant, 2023-05-01:
This has bothered me since I started using Cura.
You can try to fix this yourself.  Go to the folder "C:\Program Files\UltiMaker Cura 5.3.1\share\uranium\plugins\Tools\RotateTool" and open the file "RotateTool.py" in a text editor like Notepad.
Line 49 is:
"        self._snap_angle = math.radians(15)
If you change the 15 to 0.5 then you will have 1/2 degree increments instead of the standard 15 degree increments.  That is usually close enough for government work.

truegret, 2023-05-01:
Hi Greg -

I appreciate the response.  However, my challenge is not the resolution, rather having to use a mouse to control the position and it takes time and rarely, if ever, am I able to get the required position.

I'll take a look and give your suggestion a try but believe a text dialog box with keyboard entry would be more helpful.

TheShanMan, 2023-05-22:
@truegret See https://github.com/Ultimaker/Cura/issues/2779.

@GregValiant thanks for that idea. Not ideal, but it'll help me quite a bit! Thanks!