Ultimaker/Cura #15409: maintenance_required flag only shanged at boot

mbhs-scott, 2023-05-02:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

On the UM5 Management Console, mark all maintenance tasks as performed

### Actual results

the maintenance_required flag on the api does not change from true to false unless you reboot the printer

### Expected results

As the devices generally stay on for a long period of time, it would be good to see the flag update the api calls live, so that monitoring dashboard will show current data, rather than needing to power cycle the machine as well

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://github.com/Ultimaker/Cura/blob/a713b1e9e979dc4e1817e43ab7a6f2a13ee3da39/plugins/UM3NetworkPrinting/src/Models/Http/ClusterPrinterStatus.py
        Creates a new cluster printer status

I took this to mean these status are created at boot. I have just tested it with one of our devices and it went from true to false after a reboot, as tasks were performed earlier in the week

MariMakes, 2023-05-08:
Hey @mbhs-scott,

Welcome to the UltiMaker Cura Github 🚀 
What an interesting report 🤔 🤩 

This is the type of feature that lives between two of our development teams.
I'll contact them both to see what they can do to improve it. 

Fingers crossed 🤞

MariMakes, 2023-05-10:
Hey @mbhs-scott,

I discussed your issue with the Cura team and they agree it's not a Cura issue. 
As mentioned in https://github.com/Ultimaker/Cura/issues/15410 it's better to make a support ticket or start a new post on [the UltiMaker Community Forum](https://community.ultimaker.com/), they are quite active there as well. 

Since it's not an issue in Cura, I'll be closing this ticket.

closed 2023-05-10 by MariMakes

mbhs-scott, 2023-05-10:
thanks I guess, it's a shame as the community directed me here, and the support team gave me a flat no because the firmware is old, regardless that the new firmware contains bugs

disappointing there is a lack of cohesive support for such an expensive device