Ultimaker/Cura #15413: (5.3.1) Cura 4 project files not opening correctly in Cura 5.3.1

DrCeeVee, 2023-05-02:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Use Cura 5.3.1 to open a project (.3MF) file created in Cura 4.x.x, which contains changes to several settings.

This has been tested with project files containing changes to the Top/Bottom settings, as illustrated in the example below.

### Actual results

Only **_some_** of the project settings are imported in Cura 5.3.1.

### Expected results

**_All_** project settings should be imported in Cura 5.3.1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

As an example, see the screenshots below, showing the imported settings of the exact same Cura 4 project file, when opened in Cura 4.13.1 and Cura 5.3.1. The imported settings in Cura 4.13.1 are correct and complete, but in Cura 5.3.1 they are **_not_** complete! **_Some_** of the settings have been imported, but **_not_** all. In Cura 5.3.1, the Top/Bottom settings still have the default values (1.05 mm, 7 layers) for the Visual profile, instead of the updated values (1.5 mm, 10 layers) contained in the project file.

[image]
[image]

MariMakes, 2023-05-09:
Hey @DrCeeVee,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

At UltiMaker we have a team that works on improving the print quality with each Cura release. 
That also includes updating printing settings for example for faster support or improved surface quality like this:
[image]

I can see that the top/bottom settings have been updated in an effort to improve the initial layer settings and reduce elephants foot: https://github.com/Ultimaker/Cura/commit/27d56d378e6e6a8e18c20ed3973a47842c7530dd

A project file only saves the settings that have been changed by you as a user. So what you are seeing is intended behavior. 

If it's important for you to save all your settings and you don't want to benefit from the research to improve your print quality there are ways to save all the settings that you used in a print.  

You can use a plug-in like this one to export all the settings to an HTML file and create a profile with all the settings. 
https://marketplace.ultimaker.com/app/cura/plugins/5axes/HTMLCuraSettings

Or if you have a recent version of Cura your UFP should contain the metadata of all the settings that have been sent to the slicer. If you save your UFP to a zip, you can find in the Cura folder a json file called slicingmetadata. You can make a new profile based on those settings. 

Can you share how this changed setting influenced your printjob? 
Would you like the team to reconsider the change that was done here?

DrCeeVee, 2023-05-10:
Just to add to my comment above, that **_I'm talking about the parameters that were changed by the user_**, not about every parameter in a profile. So, if a profile (Default, Visual, Engineering, etc.) in new versions of Cura contains improved parameter values that make better prints, but these parameters were not explicitly modified by the user and stored in a project file, then these parameters can take their new, improved values, so as to take advantage of the improvements in the profile.

But, as I said in my comments above, if a user **_explicitly changes some parameters_** that affect the structural strength (or any other property) of a part, then these parameter changes should be imported when the project file is loaded, regardless of what version of Cura is used.

For example, if a part **_must_** have very thick top/bottom surfaces (because it needs to be strong), and a project file is saved containing the increased top/bottom surface thickness parameters, then that project file, when loaded in a new version of Cura, should print with the same (increased) top/bottom surface thicknesses. Otherwise, saving a project file becomes pointless.

I hope this additional comment further clarifies the issue.

DrCeeVee, 2023-05-10:
Hello @MariMakes,

I've just noticed something that may point to the source of the problem!

If you see my Cura 4.13.1 screenshot, the Top/Bottom parameters **_do not_** have the blue circular arrow icons (that indicate a parameter change). This is strange, since the Visual profile's default values are not the ones shown! The missing arrow icons means that the Top/Bottom parameters are considered as unchanged! If this is the case, then Cura 5.3.1 correctly reverts to the default profile values (1.05 mm, 7 layers), instead of the user-changed values (1.5 mm, 10 layers).

I suspect that the problem may be with my project files... Maybe there was some corruption that resulted in some parameters being marked as "changed" while others being marked as "unchanged" (although they **_were_** changed by me). This is good news, because it means that I will just have to redo the bad project files and everything should then work as intended.

If the above is true, then everything should be fine with Cura 5.3.1 and I will need to test my project files for this. I will do some tests back home, where I have both Cura 4.13.1 and 5.3.1, and let you know here.

Thanks again, will be back with further results later on today.

DrCeeVee, 2023-05-14:
Hello @MariMakes and everyone!

I finally found some time to do the tests I promised in my last comment. It turns out that the problem was with the project files themselves, and **_NOT_** with Cura 5.3.1. For some odd reason, some of the parameters I had changed in Cura 4 were flagged as "not changed" in the project file, and Cura 5.3.1 simply didn't update them when the project file was loaded. I don't know what caused the project files to behave like this, perhaps a bug in an older version of Cura 4 I was using at the time.

I have now updated all my Cura project files, and they all load correctly in Cura 5.3.1.

In conclusion, there is no need to do anything in Cura 5.3.1. I will close this issue.

Thanks to @MariMakes for taking the time to respond, and please accept my apologies for raising an issue which was not what I initially thought it was... I'm happy that my confidence in Cura 5.3.1 has been restored and that my project files are now correctly loading.

closed 2023-05-14 by DrCeeVee

MariMakes, 2023-05-15:
[image]

That's great news! 
I did show your problem to the team but they were also at a loss for how things could be flagged as "not changed".
If you are interested, a setting can come from many places [you can read more about it on our wiki.](https://github.com/Ultimaker/Cura/wiki/Container-Stacks)