Ultimaker/Cura #15414: Artifacts in walltoolpar

ganeshmev, 2023-05-02:
### Is your feature request related to a problem?

[image]
The above change in print speed gets printed and is seen as an artifact on the printed part.
Where is it occurring?
[image]
Below is the side where the issue is seen
[image]
Because of the change in geometry, This is seen. 

### Describe the solution you'd like

A way to compensate the wall flow with Cura 5.x wall compensation. 
[image]

Below is the output from Cura 3.x
[image]
No artifacts seen.
[image]
 

### Describe alternatives you've considered

-

### Affected users and/or printers

All machines listed

GregValiant, 2023-05-04:
You can see in the Cura 3 image that there are three extrusions to the wall but in the Cura 5 image there are only two.  As Cura alters the line width to make up for the missing extrusion line; segments are added to the line because the flow changes.  There is also a speed alteration to maintain a flow rate.
You can try adjusting "Flow Rate Compensation Factor" to 0%.  You could also try adjusting the Minimum Line Width closer to your Line Width.

ganeshmev, 2023-05-05:
Thanks for reply. Appreciate the effort.
I did try to make Flow Rate Compensation Factor as Zero, 
Result is below
[image]
[test_part.zip](https://github.com/Ultimaker/Cura/files/11404903/test_part.zip)

Minimum wall Line width too doesn't solve. 
I have attached the part which am using as well

GregValiant, 2023-05-05:
Load the part into Cura and use the "File | Save Project" command.  Zip the 3mf file and post the zip folder here.

When I slice your STL with a .4 nozzle I get this in the "Speed" view:
[image]
This is with a 0.8 nozzle.  You can see that this is going to leave a shadow of the pocket on the backside.
[image]

Post that project file.  This may be a bug or a settings issue.  The Line Width seems to be having a definite effect into it.

ganeshmev, 2023-09-26:
@GregValiant Please review the zip.
[JE_test_part.zip](https://github.com/Ultimaker/Cura/files/12723436/JE_test_part.zip)
I get the below.
I get this with a 0.4mm nozzle at 200 microns layer height.
[image]
I get this with a 0.6mm nozzle at 200 microns layer height.
[image]
I get this with a 0.25mm nozzle at 200 microns layer height.
[image]

May be there needs some development which needs to happen in handling of thin walls.