Ultimaker/Cura #15415: Printers

Markr1964, 2023-05-02:
### Is your feature request related to a problem?

Can you please add Sovol printers ? I have the Sovol SV04 Idex Printer and cannot seem to make  it work with this program .

### Describe the solution you'd like

Add Sovol Printer SV04 and profiles for such.

Thank you.

### Describe alternatives you've considered

none

### Affected users and/or printers

all SV04 printers

### Additional information & file uploads

None

Markr1964, 2023-05-02:
Appreciate Feedback .

spqr2001, 2023-05-05:
thumbs up, would be really nice to have sv04

fieldOfView, 2023-05-05:
We rely on contributions from printer manufacturers or people in the community for all of the non-UltiMaker printers that are in Cura. The main reason for this is that we don't have all of the printers that are in Cura, which makes it extremely hard to provide meaningful settings. 

If you would like to contribute a profile, we would gladly welcome the contribution. You can find more information on how to do that here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

For example, the SV06 Plus was added like this:
https://github.com/Ultimaker/Cura/pull/15192

closed 2023-05-05 by fieldOfView