Ultimaker/Cura #15417: Problem slicing following shape

chronos99, 2023-05-02:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3 S1

### Reproduction steps

1. Designed the following spacer on Tinkercad. Downloaded and opened it in Cura
2. Pre slicing the shape of the spacer looks correct but when slicing the shape is incorrect

### Actual results

Slicer does not render shape correctly and printer does not print shape designed  

### Expected results

Should create a small spacer made with PLA

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Spacers.zip](https://github.com/Ultimaker/Cura/files/11375862/Spacers.zip)

GregValiant, 2023-05-04:
Thanks for the report.
I you load the model(s) and use the "File | Save Project" command the 3mf file will contain your models, the Cura settings, and your printer.  Without the models we can't really tell what might be going on.

GregValiant, 2023-05-04:
Thanks for the file.
Number of Walls is set to "1" and that's because Spiralize is turned on.  By default, if Spiralize is turned on then "Remove all Holes" is also turned on.  Additionally, you have "Use Adaptive Layers" turned on and that isn't required for this part.  All it's doing is increasing your layer height for most of the model to "Layer Height" + "Adaptive Layer Maximum Variation" so your layer height is actually 0.20 for the print
This is your model with Dynamic quality and with those settings adjusted:  The hole goes through now and the layer height is 0.16.
[image]

Is that what you were looking for or do you want those settings to be active?  Because if they are active then you get what you got.
You can make changes to the built-in quality profiles (or really to any profile), but your changes aren't permanent.  When you click on the profile name you will get the choice to "Discard" or "Keep" your customization.  So to reset a profile just select Discard and it will go back to its defaults.

Let us know if that solves your problem, or if maybe I'm looking at this wrong?

closed 2023-05-30 by github-actions[bot]