Ultimaker/Cura #15418: "Infill Line DIrection" option missing

macleancampbell, 2023-05-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Attempt to change "Infill Line Direction"
2. Do not find option available anymore

### Actual results

"Infill Line Direction" does not exist anymore. Cannot change the direction of infill pattern anymore.

### Expected results

I expected to have the option "Infill Line Distance" still in Cura, but it's not. I cannot change the direction of the infill pattern, and I am required to do so, so I am fully stuck now.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No other notes, I have explained everything already. No attached files necessary, as it has nothing to do with the model, the stl, or anything else, except the core program of Cura.
[this is a blank text file.txt](https://github.com/Ultimaker/Cura/files/11377089/this.is.a.blank.text.file.txt)

GregValiant, 2023-05-02:
Thanks for the report.
This is why we ask for Project Files.  I have no problem with this as you can see the setting right there in my screenshot.  Not all infill types use that setting and if they don't (Cross3D is one) then it becomes hidden, but it is there for those that do utilize it.
[image]

I'll remove the bug label as this seems to be more of a mis-understanding.  Let us know if you feel that is premature.

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant