Ultimaker/Cura #15420: Print Begins to Step/Skew/Shift

ajokela, 2023-05-03:
### Application Version

5.3.1

### Platform

Ubuntu 20.04.6 LTS - 5.15.0-71-generic #78~20.04.1-Ubuntu SMP

### Printer

Creality Ender 3 Pro

### Reproduction steps

Sliced any large footprint model using default "standard quality"
Upload to Octoprint
Print

### Actual results

Anywhere from a centimeter to two centimeters worth of layers, subsequent layers began to shift along the Y-axis.  If I rotated the model 90 degrees and sliced, the shifting occurred on the X-axis.

### Expected results

The print would not begin to step part way through the print job.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_MantaE3EZ-box.3mf.zip](https://github.com/Ultimaker/Cura/files/11378776/CE3PRO_MantaE3EZ-box.3mf.zip)

GregValiant, 2023-05-04:
Thanks for the report.
I can't duplicate this.  The gcode I generated from your project file is correct.  To take Cura out of the loop I read the Gcode into AutoCAD and it showed no shifts anywhere.  My printer is also an Ender 3 Pro and I have never seen any shifting from a Cura gcode.
If you open the gcode in Cura - how does the preview look?  The gcode reader is a separate app and has no connection to the slicer.  A gcode preview is a true and accurate representation of the toolpaths contained in the gcode..  If your print is turning out different than what the preview of the gcode shows then something else is going on.  It could be the printer/processor or a mechanical problem with the printer, or Octoprint might be doing something odd.
I'll leave the bug label on this and the Cura Team will take a look.  Something to consider is that if this is a Cura bug then it's the sort of bug that is serious enough that nobody would use the program.
Since the problem seems to move when you re-orient the model on the Cura build plate that makes it difficult to de-bug but I didn't find anything in the gcode to account for the behavior you are seeing.

GregValiant, 2024-06-22:
I'll go ahead and close this as it appeared to have been a print problem.

closed 2024-06-22 by GregValiant