Ultimaker/Cura #15422: Outer Wall Inset: Causing protruding outer wall artifacts in many models

dgrat, 2023-05-03:
### Application Version

5.*.*

### Platform

Windows

### Printer

All

### Reproduction steps

Using a 0.6mm or 0.4mm nozzle and the appended model will cause surface artifacts which are not visible in the gcode preview.
In the example below I defined an inset of 0.1mm using a 0.4mm nozzle while printing @100/75 mm/s on a Delta with LinAdv set to 0.22s.

### Actual results

[image]
[inset_issue.zip](https://github.com/Ultimaker/Cura/files/11379317/inset_issue.zip)

With and without inset :( Artifacts are all over the print, but I made a most prominent photo..

### Expected results

This setting should either be moved to experimental, because it is poorly implemented or fixed in a way, that at least outer walls have no protrusions.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_inset_issue.zip](https://github.com/Ultimaker/Cura/files/11379350/CFFFP_inset_issue.zip)

Sab-tech-lab, 2023-05-03:
I have the same problem with a small print, i have tried with prusaslicer without Arachne Engine enabled, and the slicing are better. I have atached e stl file for slicing. Thanks for the Arachne Engine!
[image]
[image]
[image]
[image]
[skoda_isofix_v15.zip](https://github.com/Ultimaker/Cura/files/11383847/skoda_isofix_v15.zip)

Sab-tech-lab, 2023-05-17:
I honestly did not intend to make a comparison between Cura and Prusa but only to report a bug in the Arachne engine, which in some situations does not seem to use the flow correctly with certain geometries.

dgrat, 2024-02-10:
Is there any update on that topic?