Ultimaker/Cura #15430: Object floating are still sliced on the build plate

TChabaa, 2023-05-03:
### Application Version

5.3.1

### Platform

Window 10

### Printer

Ender 3 V2

### Reproduction steps

1. Dropping an Object onto the buildplate.
2. Setting x to 2
3. Hit "Slice"
4. In Preview you can see the object is still sliced on the build plate

screenshoot
[image]

### Actual results

Print is sliced onto the build plate

### Expected results

Print should be sliced "mid air"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

"Automatically drop models to the build plate" is disabled under settings.
"remove empty first layer" is unchecked.

[image]
[image]

GregValiant, 2023-05-03:
Thanks for the report.
This is a duplicate and I'll mark it as such.  Here is what to do.

- Turn off "Generate Support".
- Go to the Mesh Fixes and turn off "Remove Empty First Layers"
- Turn on "Generate Support".

The setting "Remove Empty First Layers" gets hidden when Support is enabled, but support looks at the setting and if it is enabled it drops the gcode to the build plate.  The result is that it over-rides the general setting "Automatically drop models to the build plate".  The fortunate thing about this bug is you can get around it now instead of having to wait for an update.

MariMakes, 2023-05-09:
Hey All,

Quick update from our side.
This issue should be resolved with the next release.
The fix has been made here: https://github.com/Ultimaker/CuraEngine/commit/3b1f2e612534f0ed05e022159a5308a6d2c30479

Thanks for the reports ❤️
Together we make Cura better 🤝