Ultimaker/Cura #15431: problem with helmet of do2d

djmathon, 2023-05-03:
### Application Version

5.3.0

### Platform

Mac

### Printer

Creality CR-10S pro

### Reproduction steps

If I want to print a helmet from Do3D something strange happens in Cura where the helmet is not wearable and in other programs this does not happen 
as you can see on the pic
[image]

### Actual results

I've tried all sorts of things
like contacted d02d but they don't really respond
Other slicing program And it doesn't happen at that 

### Expected results

tell me what I am doing wrong or how I can fix the problem

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

let me know if you need sorting

GregValiant, 2023-05-04:
Take a look in the Mesh Fixes section and make sure that "Make Overhangs Printable" is NOT checked.  Also make sure that "Remove all Holes" is NOT checked.  Both of them are specialty settings and should be turned off by default.  "Remove all Holes" gets enabled when Spiralize is turned on but it should disabled when Spiralize is turned off.

MariMakes, 2023-05-08:
Hey @djmathon,

Thanks for your report 👍

I really like the suggestions from @GregValiant.
Can you check if updating the following settings will show the behavior you are looking for:
1. "Make OVerhangs Printable"
2. "Remove All Holes" 

Can you let us know if that fixes your issue?

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-28 by github-actions[bot]