Ultimaker/Cura #15433: Layer cutting problem

rybianets, 2023-05-03:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Anet A8 plus

### Reproduction steps

About slicing the model in the middle of the model, support layers are generated.
Loaded the model and performed cutting.
Supports included, 60% infill.

### Actual results

[image]
[3D Objects.zip](https://github.com/Ultimaker/Cura/files/11389652/3D.Objects.zip)

### Expected results

Cura 4.6
[image]
.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Attached is project and model.

GregValiant, 2023-05-04:
Thanks for the report and the files.
Looking at the preview with supports turned off you can see that there are gaps in the preview where there shouldn't be any.  That indicates errors in the model (usually flipped normals).
Here is your model on the left and a model I repaired using MS 3D Builder is on the right.  You would need to check to make sure but it would appear that the problems have gone away on the repaired model.
[image]

I'm going to remove the bug label from this as it appears to be a model problem.  I hope you understand.
If you feel that is premature let us know.  I'll leave this open so you can respond here.

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant