Ultimaker/Cura #15434: Error Slicing for Dremel 3D40 and 3D45

KarlTheStudent, 2023-05-03:
### Application Version

5.3.1

### Platform

MacOS Monterey, Chip: Apple M1

### Printer

Dremel 3D40 and 3D45

### Reproduction steps

1. Load a stl file
2. Supports and Raft on
3. Click Slice
4. Got an "unexpected error"

### Actual results

unexpected error

### Expected results

The slice file should be available as a .g3drem
[image]
[dremel slice error.3mf.zip](https://github.com/Ultimaker/Cura/files/11390113/dremel.slice.error.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11390642/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11390650/cura.log)
[dremel slice error.3mf.zip](https://github.com/Ultimaker/Cura/files/11390651/dremel.slice.error.3mf.zip)

GregValiant, 2023-05-04:
Thanks for the report and the files.
The model has some errors  but they appear to be minor.  I didn't have any trouble slicing your project.
I'll leave the bug label on this.  You might try repairing the model to see if it makes a difference on your MAC system.
This is from your log file:
_"Failing to discretize parabola! Must add an apex or one of the endpoints."
"Backend exited abnormally with return code -11!"_

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic