Ultimaker/Cura #15435: Feature to Auto Space Models on Simple Grid Pattern

MadFDMer, 2023-05-03:
### Is your feature request related to a problem?

I print multiple copies of a single item, and print using "one at a time" method. So I would like to Cura to auto space the models in a simple grid pattern so they are touching at the bounds of the extruder exclusion area (gray area around model). See picture attached of what I mean. I did this manually, would like Cura to do it in 1 or 2 clicks.
[image]

### Describe the solution you'd like

Would like to be able to select the model I want multiplied, then have the option(s) for Cura to do:
- Say I want 12 copies, in 3 rows and 3 columns, Cura do quick auto spacing to maximize or minimize space between models.
- Say I want 12 copies, 3 rows and 3 columns, and I want row spacing to be X.XX mm, and column spacing to be X.XXmm
- Say I want XX copies, just fill the build plate and maximize space between each model evenly
- Say I want XX copies, just fill the build plate and minimize the space between each model (gray exclusion zone boundaries are touching each other).
- Say I want to fill the build plate with maximum copies that will fit.
- (A bit more advanced feature, not as important as above features) Say I have 10 different models on the build plate, if I select all the models, then I should be able to use some of the grid spacing options above to auto space all the models in a grid pattern (ie, space the models in a grid pattern with with minimal spacing (boundary zones touching).

(All options would be a simple grid pattern and entire pattern would be centered on build plate)

### Describe alternatives you've considered

I have to do it manually, and it is time consuming and tedious. Should be very simple to implement.

### Affected users and/or printers

Everyone that prints multiple pieces using "One at a Time" method.

carribeiro, 2023-06-23:
I'd use it. How can I upvote it?

MadFDMer, 2023-06-23:
Click the smiley face at bottom of post and select the thumbs up.