Ultimaker/Cura #15436: Print Settings change when switching printers

MadFDMer, 2023-05-04:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality CR-6 Max

### Reproduction steps

I have two Creality CR-6 Max printers setup in Cura, both with similar settings. When I switch from one of the printers to the other, the Print settings change (see video attached). In the video the Build Plate Adhesion settings change. And they will change back if I switch back to the other printer. The same print settings profile is still selected in both cases. When I switch printers it seems to change the prints settings back to the default settings in the profile, and back and forth depending on which printer is selected.

https://user-images.githubusercontent.com/131235177/236082911-4259bac2-23c6-47b4-9cc1-f7d0ff6860b6.mp4

### Actual results

The print settings change, and they should not change just because I switch printers.

### Expected results

The print settings be unaffected by switching printers.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

See video attached.

GregValiant, 2023-05-04:
Thanks for the report.
My understanding is that at least some of the settings are are saved "per printer".  They can be over-ridden by re-selecting the Setting Profile from the list, but each instance of a printer can be different.  I have 3 Ender's installed although I only have the one printer.  In my case I wanted different StartUp Gcodes and nozzles depending on what material I am printing.
So what you have may be "expected behavior" rather than a bug.  See if re-setting the profile gets them both synced up again.
The material choice can also have an effect.  I'm not sure which is the "Top Over-Ride" (material, profile, or printer) but they are certainly intertwined somehow.

MadFDMer, 2023-05-04:
Hi Greg,
I opened the same project (.3mf) file again and tried again, but this time it didn't do it. I changed the printer and the print settings were unaffected (as I would expect). So I'm pretty sure it's a bug, but now I'm not sure exactly what is causing it. It has happened to me several times and I have messed up several prints because it changed settings without me knowing because I changed the printer. Next time it happens again I will try to figure out exactly what I did that causes it to happen. I'm thinking right now it has something to do with when I modify settings in the print profile right before I print.

MadFDMer, 2023-05-21:
Hi Greg,
So I figured out what does it. FYI, I updated to 5.3.1 and it still does it. So basically you need at least 2 printers setup, and 2 print profiles setup. Create a project .3mf, and save it with Print Profile #1 using Printer #1. Close Cura. Open Cura, change to Printer #2 and Print Profile #2. Close Cura. Now open the project file and update existing when asked. After file is open, switch to Printer #2, and Cura will also change to Print Profile #2 without asking or notifying you and you will lose all your print settings that were saved in the project file. They will come back if you switch back to Printer #1. So basically, if you switch printers after opening a project file, it will change to the print profile that Cura last had open with that printer.  So I would say this is a bug...If I have multiple printers or a print farm, when I open a project I need to switch to my available printer and just print away without losing all my print settings. It just can't be doing this, I can't think of any reason why anyone would want Cura to act like this. At the very very least it should ask or notify or something that it is switching profiles. And like I said, I messed up many prints because I didn't know it was doing this and I thought I was going insane pulling my hair out because I swore that I saved all my files.

GregValiant, 2023-05-21:
The bug label is still on this so the Cura team will investigate.
I have customized Cura for my needs.  That includes printer definition files along with settings profiles.  I open a lot of other peoples project files and I have to constantly watch to make sure that when I open an Ender 3 Pro project that I don't over-write something (StartUp and Ending Gcode come right to mind) that I don't want over-written.
When a project file is opened it should pass over all the Cura settings,  the model, and the printer as they were set when the project file was created.   The other option (Models Only) leaves the settings and printer alone.  That's normal.  I would be severely hampered investigating these reports if it worked any other way.
When you "over-ride" the settings of a profile is when things get sticky.  The Setting Profile itself is a set of over-rides to one of the stock profiles.
There are bug reports here that note that changing material doesn't change current settings.  But if those settings are over-rides of the active Setting Profile then what takes precedence - the over-ride that the user asked for, or the over-ride of a material change that the user asked for?  When do you over-ride the over-ride?

I'm just saying that it isn't easy.  Everyone has their own preferences for the "Material,-Settings-Printer" hierarchy.  My work-around is to basically ignore the material.  The difference between Generic PLA and Matter Hacker PLA is insignificant.  Much more important is how "Greg's Ender 3 Pro" prints the PLA I have sitting here (which is nothing like printing TPU or PETG).
One thing is certain --- ya gotta pay attention because there is no "Easy" button.