Ultimaker/Cura #15437: cant access full print settings no matter what

WM-Wil, 2023-05-04:
### Application Version

5.3.0

### Platform

windows 11

### Printer

ender 3v2 

### Reproduction steps

i cant access full print settings even when i check all options in settings, i only visualize basic options

### Actual results

i cant access full print settings even when i check all options in settings, i only visualize basic options

### Expected results

i cant access full print settings even when i check all options in settings, i only visualize basic options ( i what to see full options)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i cant access full print settings even when i check all options in settings, i only visualize basic options

GregValiant, 2023-05-04:
Thanks for the report.
There are "Basic" controls and if you move the slider to "Custom" you get access to the full set of controls.  Even with "access" you may need to use the button to the right of the "Search Settings" box (setting visibility tool) and set the visibility to "All".
Let us know if that works for you.

WM-Wil, 2023-05-07:
Thanks you. I don't know why but the option wasn't there. I reinstalled the
program and now it's working :)

MariMakes, 2023-05-08:
Hey @WM-Wil,

Thanks for your update👍
I'll close the ticket now your issue is resolved. 😊 

Happy printing!

closed 2023-05-08 by MariMakes