Ultimaker/Cura #15439: Slicing Failed with an Unexpected Error

GGriff13, 2023-05-04:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Loaded model, added support blockers to halve line widths on detailed sections, slice. Same process I've used on more complicaated models. Tried moving support blockers slightly and offsetting down 0.5mm.

### Actual results

Error with no error code or information.

### Expected results

Sliced properly.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11396421/cura.log)

GregValiant, 2023-05-04:
Thanks for the report.
This is from the log file: "Backend exited abnormally with return code 3221225477!"
My understanding is that is a memory error.  If you could post a Project File it would help to de-bug this.

GGriff13, 2023-05-04:
[CE3E3V2_Adamandias.zip](https://github.com/Ultimaker/Cura/files/11397664/CE3E3V2_Adamandias.zip)

Hopefully that works. The support blockers work when I have the per model setting set to infill mesh only but as soon as I change it to cutting mesh it will give me the error, regardless of what settings I choose.

GregValiant, 2023-05-04:
That'll work.
Here is the model in Mesh Mixer.  The magenta and red balls indicate errors.
[image]

After using MS 3D Builder to repair the model I still could not get it to slice with the mesh modifiers in place.
Someone from the Cura team will take a look.