Ultimaker/Cura #15442: user/appdata map not cleaned

j-snijder, 2023-05-04:
### Application Version

5.2.0 / 5.3.0

### Platform

Windows 11

### Printer

Ender 3pro

### Reproduction steps

Installed Cura 5.3.1
Got message that profiles were damaged due to zonemaster printer
Uninstalled zonemaster and checked user profile folder in C:\Users\xxxx\AppData\Roaming\cura\5.3\user
Uninstalled printers still in mag and NOT deleted properly after deletion from Cura.

### Actual results

Uninstalled printers still in mag and NOT deleted properly after deletion from Cura.

### Expected results

When deleting a printer from Cura I expect All instances in each configuration file to be deleted and removed from Cura,

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No logfiles are needed for this problem afaik

MariMakes, 2023-05-08:
Hey @j-snijder,

Thanks for your report 👍
Are you experiencing pop-up messages like this?
[image]

That one often pops up if something isn't removed or upgraded correctly. 

There is a nice solution to that. You can install the start-up optimizer plug-in. 
You can download it here: https://github.com/fieldOfView/Cura-StartOptimiser

Can you let us know if that resolves your issue?

j-snijder, 2023-05-08:
Yep,, thats the message.I let Cura reset all, remove all my profiles, then after setup rollback an old backup where the printer wasn't in ut and solved it that way.And btw, its not a question of wrongly removing something but cura not accepting a manually added printer profile of the zonestarbprinters, which offcourse cura has not standard in his data. That regarding, cura is not even close to beeing updated on the market 😁

MariMakes, 2023-05-09:
Hey @j-snijder,

Do I understand correctly that your issue is resolved? 
Would you still like to keep this issue open or do you agree that we can close it?

j-snijder, 2023-05-09:
Well solved is a big statement, I had to fully let Cura remoove all my profiles, so No not solved but had no other choise as to delete all profiles and download an old backup.

Pleas look into it and make Cura fully deleted old profiles if the user removes them, letting them ly around in an unused state is sloppy and uses up unnessecary space.

MariMakes, 2023-05-09:
Hey @j-snijder,

Did you try the [Cura-StartOptimiser](https://github.com/fieldOfView/Cura-StartOptimiser)?
That plug-in gives you the option to remove the old profiles. 
[image]

Is that the solution you are looking for?

j-snijder, 2023-05-09:
Did not as it wasn't installed :(

MariMakes, 2023-05-09:
Hey @j-snijder,

I'm a bit at a loss. 
How can I best help you?

j-snijder, 2023-05-09:
The damage is done, so nothing to be done at it.Close this as solved please.

closed 2023-05-09 by MariMakes