Ultimaker/Cura #15444: Cura will not slice

ericspivey, 2023-05-05:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Loaded an stl file, tried to slice it with custom print settings that worked on a similar file, slicing failed.

### Actual results

Slicing fails. Repeatedly.

### Expected results

Slicing should not have failed.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

File is "JP_cranium_logo.stl" that can be downloaded here: https://www.thingiverse.com/thing:3884326
I had similar problems with a skull file that I had generated from medical imaging. I tried this one because it has been vetted on Thingaverse. The project file is too large to upload, please advise.
[cura.log](https://github.com/Ultimaker/Cura/files/11403055/cura.log)

GregValiant, 2023-05-05:
Thanks for the report.
This is why we aski for project files ("File | Save Project") because I can't duplicate this with my Ender and Cura settings.
[image]

There have been issues with S5's not slicing though. #15429 and #15386 I think.  Someone from the Cura team will take a look.

ericspivey, 2023-05-05:
Will I get any feedback from the Cura team?
I have had some success with slicing, since submitting this yesterday, and it is because I increased the layer thickness to 1 mm. This is actually okay for what I'm doing right now, but will not be okay soon.

ericspivey, 2023-05-05:
Hey! Just wanted to update you -- it turns out what totally solves my problem completely is rolling back to Cura version 4.13.1!!! Tell the Cura guys that, please. I wonder what they did to break Cura?

GregValiant, 2023-05-07:
4.13.1 remains a benchmark for Cura.  The 5.x versions incorporate numerous changes including all the dialogs and controls and changing the slicing engine to utilize variable line widths.  It has not been a smooth transition.

MariMakes, 2023-05-08:
Hi @ericspivey,

Someone from the Cura Team Here 👋 
There are a lot of reasons that you can get a slicing crash. 

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

ericspivey, 2023-05-08:
I tried all your troubleshooting tips with an earlier stl and this one, and none of it worked. I didn't try MeshTools, but my problem has since resolved by reverting to v4-13. I don't mind helping you figure out what's wrong with 5.3, but my time is limited. I made a project file for this ticket, but it's too large to upload, so if you send me an upload link, I'll give it to you. I could also send you a project file for the successfully sliced stl using v4-13 if you think that would help. My earlier ticket for the first stl was [15382](https://github.com/Ultimaker/Cura/issues/15382#issuecomment-1532472907), that project file was small enough to upload, so I did.