Ultimaker/Cura #15445: slicing issues, slicer is unusable

November1234, 2023-05-05:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Peopoly Moai 130

### Reproduction steps

tried to slice anything and it wont slice, I've used different versions even the "special" version. Ive tried rotating, placing in a different axis. nothing...... this is the only slicer that works for my printer so without this working im completely shut down. heres a model as well even though i believe this has everything to do with the slicer. by the way formware3d was the only slicer that worked perfectly for me but my trail is over.....back to cura and back to the same issue. also supports for the models in general just dont pop up for me at all my drop down box wont do anything at all so i have to use chitubox everytime for adding supports, just tried adding a 3d model but github doesnt support it so anyways this isnt a model issue its a cura issue as we can tell from all the other slicing support tickets, anyways hope to get some actual help thanks cura team!

### Actual results

.

### Expected results

slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

yep

GregValiant, 2023-05-05:
I seem to recall conflicts between Chitubox and Cura.  I never had the problem and I can't remember what the fix was.  Maybe @MariMakes knows what to do about this.

MariMakes, 2023-05-05:
Hey @November1234,

We used to have an issue with ChituBox and Cura on the same device but that should've been resolved by upgrading to a Cura version that is 5.x. 

We really want to help, but we are missing some information. The slicer not working can have a lot of different causes. 

Were you able to slice other models or was it just the one you tried to share? 
Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

If it's not related to your model, we would love to see your configuration folder. 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-29 by github-actions[bot]