Ultimaker/Cura #15446: .ufp files opening or printing on S5. shows "something is wrong with this file" on touchscreen

samori81, 2023-05-05:
### Application Version

5.3.1 also 5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Slice a .stl in Cura for the S5
2. Move file over with USB 
3. File on touchscreen has red ! and says "something is wrong with this file"
4. Tried opening .ufp file in cura and get "failed to load E:filename.ufp The file could be corrupt, inaccessible or does not contain any models." Screenshot attached to this ticket.4. Tried moving file over Digital factory cloud but file just says "in transit" till it timeout
5. Tried other .stl files I have sliced before without problem- Same error
6. Tried another USB- Same problem
7. Tried Formating new USB- same problem
8. Tried updating S5 Firmware to stable- same problem
9. Tried older version of Cura- Same problem
10. Tried another Computer- Same problem
11. tried updating S5 firmware to latest- same problem
12. tried removing "_" from filename-same problem

File sends and prints fine for Ultimaker 2 Connect that doesn't use .ufp files

### Actual results

File won't open

### Expected results

File should open

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-05-08:
Hey @samori81,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Can you help me understand why you want to open a UFP in Cura?  A UFP is just a set of printing instructions, you cannot change it anymore.  If you still want to change settings you need a project file.  To save a project file go to File -> Save project. 

There can be multiple reasons that a file can be corrupt. A common one is where you have closed Cura before the file was completely saved. Could that be the case? 

If you are in a crunch and still really need the .gcode that's inside of the UFP you can try the following steps.
1. Rename your file
2. remove the .ufp and replace it with .zip
3. Click Yes on the following pop-up
4. Open the .zip ( I use 7zip for that)
5.  Go to the folder 3D
6. Extract the model.gcode
7. Go to Cura and open the model.gcode that you extracted

I hope this helps 😇

samori81, 2023-05-09:
Hi, 
**Can you help me understand why you want to open a UFP in Cura? A UFP is just a set of printing instructions, you cannot change it anymore. If you still want to change settings you need a project file. To save a project file go to File -> Save project.**

Thanks for getting back to me. I only tried to open the .ufp on my PC to see if it would open up. To give me an idea if it was corrupted. Cura will open the non-corrupted files that I can also open on the S5 touchscreen.  The files I couldn't open on the S5 also refused to open on my PC.   

**There can be multiple reasons that a file can be corrupt. A common one is where you have closed Cura before the file was completely saved. Could that be the case?**

I always click the eject button when using the "save to removable drive" option within Cura. I haven't been closing Cura prematurely . The files also fail to send over the digital factory cloud too.

**If you are in a crunch and still really need the .gcode that's inside of the UFP you can try the following steps.

Rename your file
remove the .ufp and replace it with .zip
Click Yes on the following pop-up
Open the .zip ( I use 7zip for that)
Go to the folder 3D
Extract the model.gcode
Go to Cura and open the model.gcode that you extracted**

These instructions worked on the corrupted un-openable files. I was able to see the gcode preview in Cura when renamed and unzipped. This doesn't help me use my S5 to print though.  
Do you have any idea what is causing .ufp files to be un-openable? 

Thanks!

MariMakes, 2023-05-09:
Hey @samori81,

I'm glad you were able to extract the Gcodes. 
Do I understand you want to reprint the file?

Can you load the Gcode on a USB and load it on your printer?
Because that seems to work for me on my S5.
[image]

samori81, 2023-05-11:
Hi MariMakes,

I am able to load that .gcode file onto a USB but it won't open on the S5 like in your pic. I receive the same "something is wrong with this file" message (see pics). That same file will open as a preview in Cura from the USB stick though (see pic).
[image]
[image]
[image]

MariMakes, 2023-05-12:
Hey @samori81,

Thanks for the updates 👍  I'm almost out of things to try to resolve it. 🙈 
The team suggested to change USBs but that seems to be something you've already tried. 

Do you perhaps have the gcode for us so that we can take a look?

samori81, 2023-05-12:
Thanks crew,

I'd like to try a fresh install of cura. I've noticed even when I uninstall all versions of cura, that when reinstalled it has my custom profiles and some other data still installed. This makes me think my fresh install isn't that "fresh". Any tips?
My new versions of cura have also been doing some other weird things. Like not having a slice button, but also no preview when viewing a mesh. Rebooting the program usually fixes this. Still it's suspicious. I was able to get the S5 .ufp file working by slicing with another older untested cura installed on another windows pc. 
There is hope!

Sam

Sam

MariMakes, 2023-05-15:
Hey @samori81,

There is hope 😄 

This is how you perform a clean install
	1. Quit the application
	2. In a file explorer window, navigate to:
			My Computer/users/[user_name]/AppData/Local/cura
			My computer/Users//AppData/Roaming/cura
	1. Delete the entire 'cura' folder of your current version. Note: Cura folders of previous versions will be imported if not deleted!
	2. Open Ultimaker Cura. If it has been reset correctly, you will be greeted with the onboarding flow
	
If you do this you will use any custom profiles. If you want to keep these make sure to make a backup.
You can read more about them here: https://support.makerbot.com/s/article/1667412704080

samori81, 2023-06-06:
Hi MariMakes,

I'm still having the same issue but it now seems isolated to just one install. I was able to get a cura 5.3.1 to write successfully on another computer. I tried the steps mentioned above but I wasn't shown the "onboarding flow" screen as mentioned.  I couldn't find any cura files in "My computer/Users//AppData/Roaming/cura" or similar.  The problem computer has multiple users who may have installed cura. Is it possible this is causing a conflict?

MariMakes, 2023-06-06:
Hey @samori81,

Thanks for getting back to us 😊 
I'm glad to hear your issue has been resolved.
I'll close the issue. 👍

closed 2023-06-06 by MariMakes