Ultimaker/Cura #15449: wire frame printing

Benchickadel, 2023-05-05:
### Application Version

5.3.1

### Platform

Mac 13.2.1

### Printer

Creality Ender-3 s1 pro

### Reproduction steps

Tried to wireframe slice a cube 40mmx40mmx40mm. 
PLA settings (only change is temp to 210)
just checked wire frame.

### Actual results

Slicing failed error message.

### Expected results

Should have created a sliced file.
[CE3S1PRO_CUBE TEST.3mf.zip](https://github.com/Ultimaker/Cura/files/11407845/CE3S1PRO_CUBE.TEST.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CE3S1PRO_CUBE TEST.3mf.zip](https://github.com/Ultimaker/Cura/files/11407853/CE3S1PRO_CUBE.TEST.3mf.zip)

GregValiant, 2023-05-08:
Thanks for the report and the files.
I've never used wire printing.  Just as well since it appears to be broken.  I'll leave this for the Cura Team.

MariMakes, 2023-05-08:
Hey @Benchickadel,

Welcome to the UltiMaker Cura Github 🚀
Sorry, it took us a while to get back to you 😞

As mentioned in the release notes, it's broken in 5.3 and we are removing it because we see it's rarely used. So if you want to use it you'll have to go back to Cura 5.2 or older. You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.2.2

Can you help me understand why you wanted to use wire printing?

closed 2023-05-28 by github-actions[bot]