Ultimaker/Cura #15452: unexpected error ('Failing to discretize parabola!')

thijses, 2023-05-06:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Artillery Sidewinder X1

### Reproduction steps

exported models from Fusion360 (using '3D print' button, not 'export' button)
tried to slice model

### Actual results

'unexpected error' message box.
the entire log file should be in the '5.3.zip' file i included as well.
but this is the log line that seems to be the problem:
`
2023-05-06 12:15:36,110 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-05-06 12:15:36.109] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
`
followed by:
`
2023-05-06 12:15:37,728 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!
`

Note: first i tried to slice this model along with several others, and a duplicate of this model. In that particular scenario, it still threw the parabola error, but didn't quit the backend.

i would love to know what it is about this particular model that causes this kind of geometrical issues in Cura's backend (edges that are too sharp?, 'watertight-ness' issues?)

### Expected results

i'm pretty sure it should just slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

here is my configuration folder contents (and yes, i closed cura before zipping, to avoid in-use conflicts)
[5.3.zip](https://github.com/Ultimaker/Cura/files/11412076/5.3.zip)
here is the model i tried to slice
[problematic model.zip](https://github.com/Ultimaker/Cura/files/11412091/problematic.model.zip)

thijses, 2023-05-06:
i just exported the model in fusion a different way (regular 'export to .3mf instead of using the '3D Print' button, and that one (while looking identical to my human eye, in terms of the polygons) does successfully slice.
[non-problematic model.zip](https://github.com/Ultimaker/Cura/files/11412242/non-problematic.model.zip)

Piezoid, 2023-05-06:
Related: https://github.com/Ultimaker/CuraEngine/pull/1737

KMVPrograms, 2023-09-01:
Hello. We have the same problem with 5.3.2 Cura. Is there any expectations on fixing this problem? What is the current way of dealing with this problem?
Thank you.

MariMakes, 2023-09-04:
Hey @KMVPrograms,

Currently, we have 4 questions that resolve a significant amount of the issues. Or at least help pinpoint where the problem is. 

1. Have you tried reverting to the default settings?
2. Have you tried slicing with Support disabled? 
3. Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
4. Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

Do any of these questions help you?

KMVPrograms, 2023-09-04:
Hi @MariMakes,

Thank you for your reply. 

4. model seems to have no issues.
2. With more investigation it looks like problem is related to Tree Support. With Default support it works fine. So most likely it corresponds to this issue https://github.com/Ultimaker/Cura/issues/15229, isn't it? Does it need to work properly with Cura 5.4 version?

MariMakes, 2023-09-04:
Hey @KMVPrograms,

The new Cura 5.4 contains the fix for a number of slicing crashes related to tree support. 
Because Cura 5.4 introduced a new type of tree support. 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

KMVPrograms, 2023-09-04:
Thank you. We will try 5.4.0. Have a nice day.