Ultimaker/Cura #15458: Slicing fail

Byrondt, 2023-05-07:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

Import model. Hit slice.

### Actual results

Slicing failed. Report a bug

### Expected results

The model should have been sliced and converted to a g-code.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AKMAX_ShoulderPlatesLeft.zip](https://github.com/Ultimaker/Cura/files/11413785/AKMAX_ShoulderPlatesLeft.zip)

GregValiant, 2023-05-08:
Thanks for the report and the file.
It wouldn't slice for me either.  When I checked the models with Mesh Tools the report said they weren't watertight.  After repairing the models and relocating them on the bed the slice finished.
[image]

With your settings you are looking at a 56 hour print.  I set it up the way I would print and they were 14 hours.  I'm not criticizing your set up but I think it can be tweaked.  I did set the print speed to 75 but that didn't account for a 42 hour difference in print time.
I think I'll leave the bug label on this for now and the Cura team will take a look but since a key to getting them to slice was to repair the models it might not be a slicing bug.

Byrondt, 2023-05-08:
Thank you! I’ll look into it more. For some reason it would slice one individually, so I just sliced and printed that one twice.Thanks,Byron Turner