Ultimaker/Cura #15459: Cura crash when slicing big model

Valnights, 2023-05-07:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

1. I imported the dragon cervical file into cura
2. I selected my profile which has sliced the same file scaled down no problem
3. I scaled it up to 118%
4. Sliced

### Actual results

 I either get a cura error prompting me to file  a report or it makes my Windows 11 machine blue screen and crash

### Expected results

It should slice my file and be ready to export.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura Crash Report File_CE3E3V2_dragoncervical.zip](https://github.com/Ultimaker/Cura/files/11413927/Cura.Crash.Report.File_CE3E3V2_dragoncervical.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11413929/cura.log)

Valnights, 2023-05-07:
I have an intel 10900k cpu, 64GB RAM and an nvidia 3080ti if that's relevant. I do run 2 monitors. I read a thread mentioning flipping the model in cure in a different direction or turning the secondary monitor off but neither of those solutions worked for me.

Valnights, 2023-05-07:
I accidentally closed the issue. It's not resolved yet.

GregValiant, 2023-05-07:
Thanks for the report and the files.
I wasn't able to slice this when I opened it but after rotating the part it did slice.  I don't see any errors in the model.
This appears to be a duplicate of numerous slicing bug reports.  I'll mark this as a dup and someone from the Cura team will take a look.
[image]

Valnights, 2023-05-07:
Hi Greg, thank you for your response. It's like going to the doctor, when you try to show someone it's hard to replicate hah! I did get this error with other models from this pack and the same windows 11 blue screen halfway through the slice but it's seemingly "random". I've tried slicing this model in various upscale % and others from the pack as well as a benchy, calicat and calibration cube. I got the same error as you with the dragoncervical file as in I couldn't slice it as is scaled, I had to flip it 180 and it worked. Other files would sometimes bluescreen and most times not. I will update if I figure out a way to replicate the error consistently.

GregValiant, 2023-05-07:
I think the Cura team would appreciate any documentation they can get on duplication of the error.  This has been a headache since 5.0 was rolled out.  It's worked on constantly and I have to believe they have a ton of hours into getting this straightened out.
It's kind of funny.  I'm here every evening looking over the bug reports and it seems like I'm one of the few who doesn't have problems with 5.x.  Over half of the bug reports are actually errors in the models though.  Both PrusaSlicer and IdeaMaker have auto-repair functions that run prior to slicing.  That helps clean up a lot of errors.  Cura lacks that and although Mesh Tools is very good, it's main focus isn't really repair and it must be manually loaded by the user.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0