Ultimaker/Cura #15461: Error slicing in Cura with FLSUN V400

ThiBenn, 2023-05-07:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

FLSUN V400

### Reproduction steps

Loading stl file
Change speed
Slice

### Actual results

Started slicing then
Stopped with the error 'Slicing failed'

### Expected results

Slicing complete

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Works with Cura 5.0
Doesn't work with cura 5.2 and 5.3
[FV_Suspension 25 v4 v1.zip](https://github.com/Ultimaker/Cura/files/11415057/FV_Suspension.25.v4.v1.zip)

GregValiant, 2023-05-07:
Thanks for the report and the file.
I can duplicate the problem with your project file.  I rotated the model 90°CCW and it sliced.
I'll mark this a a duplicate slicing error as there have been numerous reports of this behavior.
Someone from the Cura team will take a look.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic