Ultimaker/Cura #15465: Consistent failure to slice complex model

WannaBAcoder, 2023-05-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Custom Voron

### Reproduction steps

Tried to slice large helmet with complex features using tree supports with settings that have successfully sliced other similarly complex models.

### Actual results

Consistently get a message about slicing failing and that I should consider reporting a bug.

### Expected results

I expect slicing to be completed successfully instead of erroring out/prompting a re-slice each time.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Something I've noticed on this error before is that shifting the model a few mm in x or y axis will sometimes resolve this by itself. Since the 5.3.0 update I've been getting this message a lot more and I think it's related to my use of complex tree supports.

Log file doesn't appear to have been recently updated, but I included it, and a few other files that were modified the time of the last instance of this message. 

Files too big to share, here's a public drive link

https://drive.google.com/file/d/1YWc-oLdW3XWfZ3lGgixbauZsZby6k2Xr/view?usp=share_link

GregValiant, 2023-05-08:
Thanks for the report and the files.
I couldn't get it to slice either.  PrusaSlicer didn't have a problem with it.
The Cura team will take a look.

MariMakes, 2023-05-08:
Hey @WannaBAcoder,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I can reproduce your error. It's quite an annoying one since it takes a long time to show up.
Thanks for sharing the file. 
[image]

I also noticed that I was getting an error message that the mesh might be faulty. 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

I'll have one of the developers take a look 👍

WannaBAcoder, 2023-05-08:
Thanks for reaching out. I can try to use some of the mesh repair tools. I
eventually got this file to slice after just retrying a bunch with tiny x/y
movements each time. I should also note this isn't the only file I've had
this happen with. I usually repair all meshes in fusion 360 before export,
but maybe something is contributing there.

Does prusa slicer support direct cura profile/project imports? I'm not
familiar with that tool.

Thanks,
Vinnie Marco

GregValiant, 2023-05-08:
I decided to play with this some more.
The helmet as posted consists of 404 separate pieces.  I used MS 3D Builder and merged all the pieces into a whole.  I brought that single piece into Cura and it took a while (maybe 30 minutes) but it did finish.  It' an 85 hour print and the gcode is 435mb.
[image]

WannaBAcoder, 2023-05-08:
Thanks for looking at this. Yeah I guess that's because of how I've had to
split all the parts up. It came in as a single .obj file with very many
parts. I imported them into fusion and combined them, but I guess it only
groups the face groups and didn't rebuild the whole mesh. I'll try this and
see if this improves anything.

I've had this happen on much simpler files but maybe there's something
going on at the file level.

Thanks!