Ultimaker/Cura #15466: Second layer fusing to skirt and brim

LazaABT, 2023-05-08:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1.Brim or skirt with a small skirt distances should be enabled
2.Slice a model with a thin wedge taken out of a bottom corner
[image]

### Actual results

Second layer is fused with skirt/brim. The bed adhesion support can only be removed destructively. 

First layer:
[image]
Second layer:
[image]
Second layer with brim instead of skirt
[image]

### Expected results

Bed support should not be fused with the model. The model outline used for build adhesion methods should be the union of at least the first two layers to prevent fusing with the model. This is especially an issue with the brim enabled, as the skirt is only an issue when the skirt distance parameter is small.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Uploaded are the project file and the model used to demonstrate the issue.
[fusing_issue.zip](https://github.com/Ultimaker/Cura/files/11416576/fusing_issue.zip)

GregValiant, 2023-05-08:
Thanks for the report and the file.
If the resolution you are using for the Z (the layer height) is too tall then yes, that small feature is going to be an issue.  When sliced at a .10 layer height the problem doesn't exist as the model can be printed as designed.  At a .2 layer height I can see that the second layer prints on the skirt/brim.
Enabling Adaptive layers with a .05 max and .025 step size the problem goes away at .20 layer height.  Additionally, a brim is not indicated for this particular part but if you were to need a brim then a Skirt with a "0" distance from the part will perform the same function (bed adhesion) but because a skirt is different than a brim the effect on the print will be different.
[image]

I don't see a bug here.  This particular situation appears to be related to settings rather than something wrong in the code.
I'll remove the bug label but I'll leave this open so you can respond.  Let me know if you feel that the bug label should be put back on.

LazaABT, 2023-05-08:
Thanks for the quick response. 
I agree that it is probably not a bug but I believe it's more of a design oversight. I could not get the result you showed by modifying layer height, enabling adaptive layers or using a skirt distance of 0 until I enabled supports. This then modifies the brim outline if there is something to support(i.e. a large enough unsupported surface which is at least 2 layers away from the bed) as the support gets added to the first layer before generating the brim/skirt. With the supports disabled this causes fusing of the second layer with the brim/skirt in all cases.
I was already aware of workarounds and the model I sent was not the model I was printing when I experienced the issue. The original model was very large covering most of the print bed surface which forced me to use a smaller skirt distance and I had taken out the wedge to help with print removal.
I realize this is usually not a big issue(wasn't in my case), but wanted to report it because it seems like an easy fix(union of the first 2-3 layers). Sorry for wasting your time if this does not warrant a fix.

GregValiant, 2023-05-08:
It's never a waste of time.  I learn something new from almost every report here and sometimes new features come from these sorts of discussions.  Ya just never know.

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant