Ultimaker/Cura #15472: Building Cura from source in Windows fails at building pyarcus.

Aurelius-Nero, 2023-05-08:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

none

### Reproduction steps

1. Simply follow the steps to build Cura. 
2. Until step 3 and then the following error occurs

### Actual results

Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
umspatial*:compiler.cppstd=20
scripta:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

umbase/0.1.7@ultimaker/stable: Not found in local cache, looking in remotes...
umbase/0.1.7@ultimaker/stable: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [1.06k]
umbase/0.1.7@ultimaker/stable: Downloaded recipe revision 86ee69dd0b8ae26b5f95f511f50b85f6
translationextractor/2.1.3@ultimaker/stable: Not found in local cache, looking in remotes...
translationextractor/2.1.3@ultimaker/stable: Trying with 'cura'...
Downloading conanmanifest.txt completed [0.06k]
Downloading conanfile.py completed [10.52k]
[... 625 lines cut ...]
pyarcus/5.2.2: CMake command: cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/b900ce/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/7566b6/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\b900ce\1\."

----Running------
> "C:\.conan\b900ce\1\build\generators\conanbuild.bat" && cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/b900ce/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/7566b6/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\b900ce\1\."
-----------------
The system cannot find the path specified.
pyarcus/5.2.2:
pyarcus/5.2.2: ERROR: Package '46f4cc8d5d301cf7f04473c21ba4cacf2020349f' build failed
pyarcus/5.2.2: WARN: Build folder C:\.conan\b900ce\1\build\Release
ERROR: pyarcus/5.2.2: Error in build() method, line 104
        cmake.configure()
        ConanException: Error 1 while executing cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/b900ce/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/7566b6/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\b900ce\1\."

### Expected results

Expecting it build properly

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

closed 2023-05-27 by github-actions[bot]

double-kill-123, 2023-09-03:
Has this defect been resolved? I encountered the same problem

TechMaverik, 2023-12-12:
I also have the same problem.