Ultimaker/Cura #15475: CI is running on draft merge requests -> Adjust triggers?

Kreyren, 2023-05-08:
### Application Version

gfdh

### Platform

sfdhsf

### Printer

fshfdh

### Reproduction steps

fdshsfdh

### Actual results

sfdhsfd

### Expected results

fsdhsfdh

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

fdhsfdh

Kreyren, 2023-05-20:
>  I'm not a developer so I don't know what best practice is in this case. -- @MariMakes (https://github.com/Ultimaker/Cura/issues/15475#issuecomment-1540142600)

In https://github.com/Ultimaker/Cura/blob/a1c01193daa5e8b76ea9ba93bdd346d572aecbdb/.github/workflows/printer-linter-pr-diagnose.yml

The types should be adjusted according to https://docs.github.com/en/actions/using-workflows/events-that-trigger-workflows#pull_request

e.g.

```diff
on:
  pull_request:
+    types; [ opened, ready_for_review, review_requested ]
    path:
      - "resources/**"
```

Assuming that the draft merge request is spamming since draft merge requests alike usually have ~20 commits before they are submitted otherwise i guess it can be kept as is.

MariMakes, 2023-06-01:
Hey @Kreyren,

Thanks for the extra comment, it was super helpful for the discussion. 
The team concluded that they prefer the tests to be run, even if it's in the draft. 
Since we have an open source project there are no financial downsides in having them run. 

We are thinking about implementing a welcome message bot for new contributors. 
Do you think this would be good to mention in there why we made the decision to keep them running? 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
I'll be closing this issue. I hope you understand.