Ultimaker/Cura #15477: import Profiles from Cura 5.3.0 to 5.3.1

TMTYD, 2023-05-08:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Anycubic Kobra Go

### Reproduction steps

I exported the profile in the old version

And want to import in the new version under profile

### Actual results

The error message invalid argument erno22 came up

### Expected results

Actually, the old profile from version 5.3.0 should be imported into 5.3.1

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[Log file](https://www.file-upload.net/download-15142038/cura.log.html)
[pla profile from 5.3.0](https://www.file-upload.net/download-15142039/pla.curaprofile.html)

TMTYD, 2023-05-12:
So I exported the profile to cura, in the folder it was then xyz.curaprofile, this file was not a zip file in the sense of xyz.curaprofile.zip.  

Yes I opened this file and edited it with a text editor to change the quality type from normal to standard.  

Since I read that.  This was also unsuccessful.  

If cura had exported the profile as a zip file, I would have opened the zip file, but that wasn't the case.  And only one file was exported per profile

MariMakes, 2023-05-15:
Hey @TMTYD,

I also thought that .curaprofiles were textfiles but I trust the developers if they say it's not.
If you want to make a change like that, and still load the profile into Cura, I made a video showing how without getting an error. 

https://github.com/Ultimaker/Cura/assets/40423138/be011f46-d902-4f14-99a6-7fc2cb469d2c

I hope this helps 😇
Can you let us know if this solves your issue?

TMTYD, 2023-05-15:
Ok thanks I can test if that works, I'll report back.  It's a pity the profiles aren't immediately exported in zip files, then this problem wouldn't be and it would be easier

TMTYD, 2023-05-15:
So I just tried that, it worked. It is enough if you add xyz.curaprofile.zip at the end, that works the same way. Thanks for the info.

As I said, it would be better if it was exported directly as a .zip file

MariMakes, 2023-05-15:
Thanks for reporting back. 
Happy to hear that it worked 😊 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
I'll be closing this issue. I hope you understand.

closed 2023-05-15 by MariMakes