Ultimaker/Cura #15481: Features protruding from a cone or cylinder are not print properly

pigaro, 2023-05-08:
### Application Version

5.3.0

### Platform

macOS

### Printer

Ender 3

### Reproduction steps

Attempted to slice and print a previously printed design. Print quality was poor. Protruding features that were previously part of the base feature (cone or cylinder) were now printed separately. A significant increase in the number of retractions were noted when compared with file sliced using Cura 4.13.1. Attached photos show results for printing design sliced with Cura 4.13.1 (desired result, wall ordering does not matter), sliced with Cura 5.3.0 with wall setting from outside-to-inside, and Cura 5.3.0 with inside-to-outside.

[image]
[image]
[image]

### Actual results

When sliced with Cura 5.3.0, protruding features are not printed with the base cone or cylinder, noted by the large number of seams in the layer preview. Results are worse when walls are printed outside-to-inside as some (but not all) protruding features are printed before the feature they attach to. Results are slightly better when walls are printed inside-to-outside as the base feature is printed before any of the protrusions.

[image]
[image]
[image]

### Expected results

When sliced with Cura 4.13.1 (and previous versions) the protruding features are printed as part of the base feature and print as expected with high quality.

One note: indented features are sliced as expected with Cura 5.3.0. Another model with the same type of feature indented on a cylinder sliced as part of the base feature.

[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Log file was not attached since Cura 5.3.0 completed slicing without issue.

Project file attached. STL or design file (Fusion 360) can be provided on request.

[large payload fairing (LPF).3mf.zip](https://github.com/Ultimaker/Cura/files/11425630/large.payload.fairing.LPF.3mf.zip)

GregValiant, 2023-05-08:
Thanks for the report and the file.
Try dropping your "Minimum Line Width" to 0.29.  That should allow an "out and back" toolpath on each of those features.   That will (should) keep the starts out of those areas.
[image]

BTW:  Your "Initial Layer Horizontal Expansion" of -0.5 is causing those little features to disappear on the first layer.  That is causing the second layer to overhang.
This is the Initial Layer with Horizontal Expansion turned off.
[image]

See if making those changes works for you and let us know.

pigaro, 2023-05-09:
Thanks. Reducing the Minimum Line Width resolved my issue. Thanks for the quick response and the explanation of why the first layer did not have the features. I had problems with elephant foot earlier and the negative initial layer seemed to help that, but I had forgotten about it. It is good to be able to use the latest release of Cura for these prints.

closed 2023-05-09 by pigaro

GregValiant, 2023-05-09:
I really liked 4.13.1 and to me it remains the benchmark for Cura.  The new "5" versions haven't caught up yet but it's getting there.  Some of the new settings in the Walls section take some getting used to.
My personal workflow is to re-set my Setting Profile at the start of every project.  Then I go through ALL the settings.  It takes a few minutes and a couple of slice previews, but in the end it saves time as I don't have to abort very often do to "%$##, I forgot to move the Z-seam!" or whatever setting I happened to have missed.

pigaro, 2023-05-09:
Greg,

Thanks for getting back to me on this issue. I was very pleased to see that your suggestion to reduce the Minimum Wall Width resolved my issue. I notice that this setting is one of the new settings for Version 5.

You wrote:
> I really liked 4.13.1 and to me it remains the benchmark for Cura. The new "5" versions haven't caught up yet but it's getting there. Some of the new settings in the Walls section take some getting used to.
> 
I guess I am a little surprised by the statement on version 5, as I thought it would have included all the work from version 4. Clearly changes have been made to the slicing algorithms as noted by new settings. The problem for me is that I have little clue how the slicer actually works, so it is sometimes hard to decide which settings to modify. When I had the problem that prompted my report I tried modifying a number of the settings but the setting that was effective is not one that I would have guessed would affect my issue. I do have the extension that explains the settings (Settings Guide), but sometimes the explanation is not entirely helpful. After reviewing the explanation, I can see how it could affect my issue but I would not have known what to do about it.

This experience has given me a newfound appreciation of Cura. Before reverting to 4.13.1, I tried another slicer and found it woefully inadequate. I am sure that if I would have invested a significant amount of time, I might be able to get it to do what I wanted, but it did not seem as professional as Cura. I was very pleased to see that I could still access older versions of Cura without having to make them from source code.

> My personal workflow is to re-set my Setting Profile at the start of every project. Then I go through ALL the settings. It takes a few minutes and a couple of slice previews, but in the end it saves [...]
I wish I had your discipline :-) Perhaps this experience will help me develop a more systematic approach to slicing (other than expecting it to automatically do what I want).

Thanks again,

Lee Mendoza
***@***.***