Ultimaker/Cura #15483: Cura beta 5.3.1 will not open.

masterfenix87x, 2023-05-09:
### Application Version

5.3.1

### Platform

windows 10

### Printer

elegoo neptune 3 pro

### Reproduction steps

Ive uninstalled and tried reinstalling cura beta and it still will not boot up. i can click on the app to open then it goes to white screen and eventually will see (not responding) message.

### Actual results

nothing, cant get into the program to do anything.

### Expected results

i would like to open cura to get some projects sliced to print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-05-09:
Hey @masterfenix87x,

Welcome to the UltiMaker Cura Github 🚀 
Sorry to hear you are stuck 😞 

If the monitor trick doesn't work. can you try one of these workarounds? 

Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

masterfenix87x, 2023-06-26:
how has this with beta 5.4 not been fixed? i shouldn't have to disconnect a monitor to use this software.

closed 2023-07-26 by github-actions[bot]