Ultimaker/Cura #15485: GCode created for 2. extruder always contains start temperature values of 1. extruder in "material_print_temperature[_layer_0]" parameters

raspstar, 2023-05-09:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

SOVOL SV04

### Reproduction steps

Setting "Printing Temperature" or "Printing Temperature Initial Layer" to different values on 2. extruder than on 1. extruder are overwritten in sliced GCode with the values from 1. extruder material settings

[image]

### Actual results

Initial M104 & M109 in GCode are wrongly set to values for 1. extruder

### Expected results

Initial M104 & M109 in GCode are correctly set to values for 2. extruder

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Sovol SV04 - Pi Zero W bracket.zip](https://github.com/Ultimaker/Cura/files/11432929/Sovol.SV04.-.Pi.Zero.W.bracket.zip)

GregValiant, 2023-05-09:
Thanks for the report.
If you would copy and paste your starting gcode here that would be good.  Because your machine is a custom job it doesn't load with the project file as I don't have a definition for it.

Right now my guess is that you are explicitly stating in your startup gcode to set the temperature, but it is not explicit which hot end you are referring to so it defaults to T0.
This gcode is with the "M109 S{material_print_temperature_layer_0}" line in the startup gcode.  I have T0 set to 205° and T1 set to 230°.
**T1**
M82 ;absolute extrusion mode
;
; Greg's E3Pro StartUp
;
G90
G21
M82
M140 S70
M109 **S205.0**
G92 E0 ; Reset E
G28 X Y; Home XY axes
You can see that it is wrong as the print will start with T1 at an incorrect 205°

This next is with the hot end temperature callouts commented out (leading semicolon).  You can see that Cura has set T0 to standby temp and T1 the "active extruder" is set to 230.
T1
M104 T0 S175
M104 S230
M105
M109 T0 S175
M105
M109 S230
M82 ;absolute extrusion mode
;
; Greg's E3Pro StartUp
;
G90
G21
M82
M140 S70
;M104 S205.0
M190 S70
;M109 S205.0

If you disable the hot end temperature lines in your startup gcode and let Cura enter them (they will go in before the StartUp gcode) then the temperatures should be correct.  The downside is if you have ABL as it is usually preferable to have the nozzles at a non-drooling temperature during leveling.

raspstar, 2023-05-09:
Hi Greg,

short: You are absolutly right.

Commenting out the relevant temperature directives solved the problem as Cura then creates the temperatures correctly before the Start G-code (1st extruder isn't used at all in this example):

###
;
;Manual Start G-code
;
;M140 S{material_bed_temperature_layer_0}
;M104 S{material_print_temperature_layer_0}
M280 P0 S160
G4 P100
;M190 S{material_bed_temperature_layer_0}
;M109 S{material_print_temperature_layer_0}
G28
G92 E0
G1 X71 Y10.0 Z0.25 F5000.0
G1 X231 Y10.0 Z0.25 F1500.0 E15
G1 X231 Y10.4 Z0.25 F5000.0
G1 X71 Y10.4 Z0.25 F1500.0 E30
G92 E0
G1 Z2.0 F3000
...

###

;Generated with Cura_SteamEngine 5.3.1
T1
M140 S60
M105
M190 S60
M104 S190
M105
M109 S190
M82 ;absolute extrusion mode

;Manual Start G-code

;M140 S60
;M104 S240
M280 P0 S160
G4 P100
;M190 S60
;M109 S240
G28
G92 E0
G1 X71 Y10.0 Z0.25 F5000.0
G1 X231 Y10.0 Z0.25 F1500.0 E15
G1 X231 Y10.4 Z0.25 F5000.0
G1 X71 Y10.4 Z0.25 F1500.0 E30
G92 E0 ;Reset Extruder
G1 Z2.0 F3000
...

###

Thanks a lot & regards,

GStar

GregValiant, 2023-05-10:
One of the upsides to Cura is that it is highly customizable but that can also be a downside because some things just aren't obvious.
Can we close this then?

raspstar, 2023-05-10:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Hi Greg,Yes & yes...Regards,GStar                                                                                                                                                                                 Von: ***@***.***: 10. Mai 2023 12:43An: ***@***.***: ***@***.***: ***@***.***; ***@***.***: Re: [Ultimaker/Cura] GCode created for 2. extruder always contains start temperature values of 1. extruder in "material_print_temperature[_layer_0]" parameters (Issue #15485)  
One of the upsides to Cura is that it is highly customizable but that can also be a downside because some things just aren't obvious.
Can we close this then?

closed 2023-05-10 by GregValiant