Ultimaker/Cura #15487: SLICING FAILED

THEDJWYLD, 2023-05-09:
### Application Version

5.2.0

### Platform

windows 10

### Printer

ENDER 5 +

### Reproduction steps

I went to slice a file and it said "SLICING FAILED" Slicing failed with an unexpected error.
Now it does this with every stl.

### Actual results

will not slice any files

### Expected results

files should be sliced.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

blakeb21, 2023-05-09:
Was having this issue on 5.1.1 and updated to 5.3.1 and am still having the same issue.

MariMakes, 2023-05-12:
Hey @THEDJWYLD (and others)

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-27 by github-actions[bot]