Ultimaker/Cura #15490: WIre Mesh - Slicer Stopped

ariberman, 2023-05-10:
### Application Version

5.3.1

### Platform

Macbook Pro (2013), OS 11.7.6

### Printer

CREALITY Ender 3 S1 Pro

### Reproduction steps

I selected Wire Printing on a relatively small model.
Two messages appeared:
- Can't display wire accurately, and,
- Slicer Stopped,
Slicer is still indicating that it is slicing (I think it's stuck.)

### Actual results

Two messages appeared:
- Can't display wire accurately, and,
- Slicing Stopped

### Expected results

Preview of the wire mesh

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No log cura.log file generated.
Can't upload the .3mf file (unsupported format).

GregValiant, 2023-05-10:
Thanks for the report.
This is a duplicate of #15450 and if you read @MariMakes reply there I think you will see where this is at.
I'll mark this as a duplicate and close this.  I hope you understand.

closed 2023-05-10 by GregValiant

MariMakes, 2023-05-11:
Hey @ariberman,

Greg is correct, it's broken and we plan to remove it in the next release.
We see that it is rarely used and is not worth trying to fix what is broken. 

If you have a good reason why we should fix it instead, you can let us know and I'll make sure it ends up on the correct desk 👍