Ultimaker/Cura #15491: Slicing Failure using SnapMaker J1 Profile, Interlocking

cbcolbeck, 2023-05-10:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

SnapMaker J1

### Reproduction steps

1. Attempting to Slice Multimaterial Print for IDEX, using Material Interlocking.
2. Slicing Fails

### Actual results

The Slicer Reports a Failure and Suggests Reporting a Bug.

### Expected results

Slicing Succeeds.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11445837/cura.log)
[38 Pair, J1.zip](https://github.com/Ultimaker/Cura/files/11445845/38.Pair.J1.zip)
[image]

MariMakes, 2023-05-12:
Hey @cbcolbeck,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

That's a pretty cool model 🤩 
Is it a type of jig? What is it for?

I took a look at your project file and I believe your error has to do with incorrect normals.
Those big red surfaces are usually an indicator of this. 

Have you tried doing a Fix Model Normals with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

That seemed to have resolve it for me.
Can you take a look if that also resolves it for you? 
If that's not it, I'll take another look at what is going wrong. 👍 

Also, can you help me understand, why do you have models placed outside of the buildplate? 
Are they there on purpose or by accident?

cbcolbeck, 2023-05-12:
Hi.
    Thanks for your reply. While I tinker with that file let me answer your questions:
    The model is a prototype pool cue rest that I've been developing, and thanks for the compliment! As for parking parts outside the build plate, I do that so I can keep all the parts of a project in one file. I hear that other slicers allow for multiple build plates in one file, and I hope that Cura will implement that feature, but this works in the meantime.
     I have some questions: I'm trying to educate myself on mesh fixes, is there a page or video on this topic that you recommend? Why can't I choose a flat surface to orient onto the build plate once I've merged a multi-part model? (I want to print my rest upside down so that the PEI texture is on the top and the stair-step surface goes on the bottom.)
     Your input prompted me to dig through my model, and I found some voids in it that must have been causing problems. Fixing the Holes and then fixing Normals seems to help too, although as I mentioned I'm still fuzzy on that concept.
     With all that fun stuff out of the way, I'm sorry to report that I'm still having slicing failures. I've been able to slice one piece of the rest or the other, but I need to be able to slice the two pieces on one plate. Sometimes, if I slice one side and then park it aside, then slice the other, then slice the pair, I get a result, though I'm not excited about the extra steps or the inconsistency.
         Thank you for your help,
             Craig

cbcolbeck, 2023-05-12:
Hi.
    Thanks for your reply. While I tinker with that file let me answer your questions:
    The model is a prototype pool cue rest that I've been developing, and thanks for the compliment! As for parking parts outside the build plate, I do that so I can keep all the parts of a project in one file. I hear that other slicers allow for multiple build plates in one file, and I hope that Cura will implement that feature, but this works in the meantime.
     I have some questions: I'm trying to educate myself on mesh fixes, is there a page or video on this topic that you recommend? Why can't I choose a flat surface to orient onto the build plate once I've merged a multi-part model? (I want to print my rest upside down so that the PEI texture is on the top and the stair-step surface goes on the bottom.)
     Your input prompted me to dig through my model, and I found some voids in it that must have been causing problems. Fixing the Normals seems to help too, although as I mentioned I'm still fuzzy on that concept.
     With all that fun stuff out of the way, I'm sorry to report that I'm still having slicing failures. I've been able to slice one piece of the rest or the other, but I need to be able to slice the two pieces on one plate.
         Thank you for your help,
             Craig

cbcolbeck, 2023-05-13:
Hi Again,
   I'm finding that a big factor in whether I can slice models with Material Interlocking is how many models there are on the buildplate. Multiple models contributes a lot to failures, and even their arrangement (such as close together or offset on the Y-axis) has an effect. I know that Interlocking is a new feature, but it's so great and I want to be able to use it for high-output projects!
   I wish that it could work on the Z-axis as well as XY. I wish I could Arrange Selected Models, rather than only being able to Arrange All. 
      Thanks,
         Craig

MariMakes, 2023-05-15:
Hey @cbcolbeck,

What a cool idea to 3D print 😊 

Regarding multiple buildplates, it was a feature in Cura 3.6 but that was sent to the shadow realm.
To quote @nallath, The implementation wasn't all that good (so it was costing us quite a bit of effort to keep it up to date). We only got a few remarks about it actually being used, so it was disabled a few releases ago. 

Regarding "Align face to build plate" does not work for grouped or merged models, we have we've added a ticket to the backlog with the intent to improve this. You can read more about it here: https://github.com/Ultimaker/Cura/issues/14237

I took another look at your project file and noticed that I can slice the project without the text on top and the rubber clamp parts. That makes me believe that there is too much detail there to use the interlocking generation on, not that there are too many models on the buildplate. 
[image]

Is it possible to reduce the complexity of these parts? Make them more boring and more geometric? 

To me, it doesn't seem like a bug in Ultimaker Cura anymore but an issue with the meshes you are trying to print.  
Would you still like to keep this issue open or do you agree that we can close it?

cbcolbeck, 2023-05-15:
> Thanks for addressing my concerns as best you can. You can close the
> ticket.
>
> Cura is a great slicer, made even better by friendly support, but at the
> end of the day my project needs what it needs, and I'm going to keep
> working until I find a way to print it as designed.
>
> Craig

Hello yet again,

With apologies for overfilling your inbox, I have some followup thoughts.

I told you that I'd get it to work! and I did. The key ended up being
editing my model so that there weren't small parts near the surface ot
the model. (In my case, it was the fillings of letters like "A" and such.)
I joined all the elements of my logos under the surface of the model so
that the Interlocking algorithm has fewer pieces to work on. Perhaps you
can suggest this to anyone else running into the issue I was having.

    Best wishes,
         Craig

MariMakes, 2023-05-19:
Hey @cbcolbeck,

That's great to hear! Thanks for the update 😊
I'll keep your suggestion in mind next time I'm troubleshooting interlocking issues. 👍 
I'll be closing this issue.  Happy Printing! 🎉

closed 2023-05-19 by MariMakes