Ultimaker/Cura #15493: Startup crash when attempting to add variant for custom printer

eqkessel, 2023-05-10:
### Application Version

5.3.1

### Platform

Linux-6.3.1-arch1-1-x86_64-with-glibc2.37

### Printer

Custom printer added to `definitons` folder

### Reproduction steps

1. Add printer definition file `makergear_m3se.def.json` to `definitions` folder based off existing files and customize, along with corresponding extruder file to `extruders` folder.
2. Add `"has_variants": true` to `makergear_m3se.def.json` file. Cura still functions at this stage, but variants dropdown just lists as "empty".
3. Create variant file `makergear_m3se_0.4.inst.cfg` in `variants/makergear` folder based on existing files and set `definition = makergear_m3se`.

### Actual results

Cura crashes on launch and reports a "Cura can't start" window with the following error traceback:
```py
Traceback (most recent call last):
  File "cura/CuraApplication.py", line 1128, in event
    return super().event(event)
  File "UM/Qt/QtApplication.py", line 502, in event
    event._function_event.call()
  File "UM/Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "cura/Settings/MachineManager.py", line 181, in setInitialActiveMachine
    self.setActiveMachine(active_machine_id)
  File "cura/Settings/MachineManager.py", line 381, in setActiveMachine
    self._validateVariantsAndMaterials(global_stack)
  File "cura/Settings/MachineManager.py", line 399, in _validateVariantsAndMaterials
    material_node = variant_node.materials.get(extruder.material.getMetaDataEntry("base_file"))
AttributeError: 'NoneType' object has no attribute 'materials'
```

### Expected results

Cura launches and variant is available for use.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11446191/cura.log)
[makergear_m3se.def.json.txt](https://github.com/Ultimaker/Cura/files/11446196/makergear_m3se.def.json.txt)
[makergear_m3se_extruder_0.def.json.txt](https://github.com/Ultimaker/Cura/files/11446200/makergear_m3se_extruder_0.def.json.txt)
[makergear_m3se_0.4.inst.cfg.txt](https://github.com/Ultimaker/Cura/files/11446211/makergear_m3se_0.4.inst.cfg.txt)

MariMakes, 2023-05-12:
Hey @eqkessel,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I took a look at your profile and I think spotted a mistake. 
These two lines seem to be missing in your makergear_m3se.def.json
    "has_variants": true,
    "preferred_quality_type": "normal",
[image]

When I've added them, I was able to slice with this printer. 
[image]

Can you let us know if this resolves your issue? Thanks 👍

eqkessel, 2023-05-16:
Can you please confirm which lines you added? The two you mentioned already exist in my definition file.
[image]