Ultimaker/Cura #15495: Cura crashes when Add Printer selected

nickr1957, 2023-05-11:
### Application Version

56.3.1

### Platform

Windows 11

### Printer

Geeetech A20

### Reproduction steps

Open Cura
Select manage Printers
Select Add Printer
Click Non-Ultimaker Printer
Select Geeetech A20
Click Add

Cura bombs out. On restart, the printer was not added.

### Actual results

Cura bombs out. On restart, the printer was not added.

### Expected results

Prinyter should appear in list of printers

This happens also on a Windows 10 machine with Vura 5.3.0 Beta

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11451719/cura.log)

nickr1957, 2023-05-11:
Should read version 5.3.1 Beta

lvanek23, 2023-05-12:
I am in same boat on my Windows 10 PC... 5.3.1 bombs out after adding a printer when i try to do anything.

Wanted to try 5.3.1 because 5.3.0 on my MacBook is leaving a horrible z seam gap

nickr1957, 2023-05-12:
Worse still, I only had to add a new one because one of my printers started causing an error on startup. Corrupt config files, only way to do anything was to reset, which deleted all my printers and profiles. Reinstalling and choosing one printer during initial setup was OK and created that printer, biut I cannot add any of the ones that were working, let alone the one that started causing isues.

Having heavily tuned settings to get perfect prints, it is all gone. 
I cannot retrieve them it seems, because there seems to be no way to use old ones in the newer versions - if I copy from 5.0 to 5.3, for example, the quality_changes folder with all the settings, I get the error about corrupt files. So I cannot retrieve the settings even for the one printer I have installed.

It looks as if,I will have to manually copy every setting across. But I can't do that for my other printers  if I can't add a printer.

MariMakes, 2023-05-12:
Hey @nickr1957,

Welcome to the UltiMaker Cura Github 🚀 
Losing custom settings is always frustrating. 😖 

The Geeetech A20 is a printer that has been updated in the 5.3 Beta and 5.3.0.
So that might be the reason that you are experiencing some trouble with the printer. 
Did you have Geeetech A20 printers before upgrading? And which version did you upgrade from? 

Also, I believe there is a chance that your profiles are still there, have you checked the Quality and User folders in your file explorer if you visit %APPDATA%\cura\<Cura version>?

The best way we can help you troubleshoot is by seeing your configuration folder. 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

nickr1957, 2023-05-12:
I have a Win10 machine on which I have Cura back to 3.4.
On this Win11 machine there is only 5.3.
The zipped config folder for 5.3 on Win11 is too big to upload - I get an arror saying there is a 25MB limit and the zip is 28MB.
So the one attached is missing the Plugins folder.
[Cura5_3ConfigWin11.zip](https://github.com/Ultimaker/Cura/files/11465283/Cura5_3ConfigWin11.zip)

On the Win10 machine I also cannot add a printer. If I run 5.2 it is fine. For 5.3 I get an error about corrupt files gor the Geeetech and if I close that and go to Printers, the Geeetech is there - if I activate it I get a fatal error and Cura crashes.

If I restart it, close the error dialog, go to Manage Printers and remove the Geeetech, then restart Cura - it still says there is a config problem with the Geeetech A20 !
Close that error again, and try to add Printer - crash as before and as on a clean 5.3 install on the Win11 machine that never had a previous Cura version.

CGessinger, 2023-08-05:
Is there any update yet? Confirmed that A20 causes Cura to crash from version 5.3 onwards

nickr1957, 2023-08-05:
I cleaned the Cura 5.3 completely from the PC in the end and installed 5.4, then created all my printers from scratch. A laborious and annoying resolution but it worked - sort of.
Cura regularly just shuts down and disappears unexpectedly. It is also hazardous to use when I am running Citrix to connect to my virtual machine in my employer's infrastructure. The PC and the Citrix screen will just go black and red grid, the explorer comes back, Cura has died and gone and I have to kill all the Citrix tasks and restart them. That never happens unless I'm running Cura.

So mostly I have learned Orcas I terrace and use that instead but there are some things I understand better in Cura so sometimes I have to use it

GHOSTS15, 2023-08-11:
i have the same problem but in my case , the printer is not connected

dgoadby, 2023-09-20:
Installed latest CURA on Win10 today and Geeetech A20 is a bust. The A10 and A20M are ok though. If I could find the profile files I could try and fix it.

dgoadby, 2023-09-21:
I have looked at the log file (attached) and the problem appears to be "Unable to find extruder definition with the id [geeetech_A20_1]"

Unlike earlier versions the printer definitions appear to be in their cloud which means that Ultimaker have to fix it. 

One fix that works is to load the A20M printer and remove the second extruder by setting "Number of Extruders" to 1.

[cura.log](https://github.com/Ultimaker/Cura/files/12679263/cura.log)

dgoadby, 2023-09-21:
I have looked at the log file (attached) and the problem appears to be "Unable to find extruder definition with the id [geeetech_A20_1]"

Unlike earlier versions the printer definitions appear to be in their cloud which means that Ultimaker have to fix it. 

One fix that works is to load the A20M printer and remove the second extruder by setting "Number of Extruders" to 1. 

I have attached a pdf of the settings page for the A20M modified for the A20. It seems to work and will be usable until Ultimaker fix their cloud version! 

These problems just highlight the problem of depending upon someone else's cloud.

[cura.log](https://github.com/Ultimaker/Cura/files/12679263/cura.log)
[A20M-single-extruder.pdf](https://github.com/Ultimaker/Cura/files/12679372/A20M-single-extruder.pdf)