Ultimaker/Cura #15500: One at a Time data insertion

GregValiant, 2023-05-11:
### Application Version

5.3.1

### Platform

Win 10

### Printer

Ender 3 Pro

### Reproduction steps

Set up for One at a Time printing.
Sliced
Attempted to post-process but the Data list was wrong.

### Actual results

This is the code when moving from one model to the next:
;TIME_ELAPSED:2478.878771

**;  DATA[102]**
G1 F2100 E304.29152

**;  DATA[103]**
M107
G0 F9000 X168.659 Y23.673 Z25
G0 X163.312 Y163.811
;LAYER_COUNT:100

**;  DATA[104]**
;LAYER:0
G1 F600 Z0.7
G0 F9000 X160.294 Y162.784 Z0.7
;TYPE:SKIRT
G1 F600 Z0.2

At every model change the data stretches away from the layer count by at least 2 and those added little snippets are hard to account for.

### Expected results

Everything between DATA[2] and DATA[len(data)-1] (inclusive) should start with ";LAYER:".  It is hard to post process and the data returned to Cura is actually in error even though that is how it was when passed to the post-processor.
;TIME_ELAPSED:2478.878771
G1 F2100 E304.29152
M107
G0 F9000 X168.659 Y23.673 Z25
G0 X163.312 Y163.811
;LAYER_COUNT:100

**;  DATA[102]**
;LAYER:0
G1 F600 Z0.7
G0 F9000 X160.294 Y162.784 Z0.7
;TYPE:SKIRT
G1 F600 Z0.2

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't think any files are required.  If I wasn't playing with post-processors I would never have noticed.  On the other hand, it is wrong and it makes debugging scripts for One at a Time a PITA (even more than usual).
My guess would be that this happens in the Cura Engine so feel free to move this report.

MariMakes, 2023-05-12:
Hey @GregValiant,

You have been busy. 😄 
Great detective work as usual 🔍 

I'll bring it up to be discussed on Monday. 
Perhaps the developers can help with your issue.

MariMakes, 2023-05-15:
Hey @GregValiant,

Quick update on our side. I did bring up your issue, but it needs more investigation from the developers. They are working hard on a deadline so I hope they can find some time to take a look after they finish their current work. 

I did notice that your remark looks kind of similar to: https://github.com/Ultimaker/Cura/issues/7980 where the conclusion was:  _It seems to be dependent on the relative position of the models. If the taller object is printed first, this bug occurs. CuraEngine is slicing the correct number of layers but the front-end is displaying it wrong._

I hope this helps 😇

GregValiant, 2023-05-15:
Hi Mari.
That's an earlier bug report regarding the layer count and it happens with a single model as well as with multiple models.  The LayerCount appears the theoretical number that could have been sliced rather than the actual layer count within the Gcode.

This bug is in regards to the way Cura handles the code for the travel moves from one model to another when in "One at a Time" mode.  Effectively, there are layers that don't start with ";LAYER:" and the inconsistency makes post-processing difficult.

rburema, 2023-09-22:
Hi Greg, sorry for the wait.

Which script are you using? Insert before/after layer? I could sort-of reproduce this when using that script. I'm not sure where the +2 comes from, but stopping the count-up should probably happen within that script itself, by parsing the number for the layer (that's the most accurate information we have).

The ChangeAtZ script for example, will parse the number after `;LAYER:` to get the layer number.

I'm not sure how to solve this in general, at least no without spending some effort on it: We can't just attach any 'layer N' to the previous one, because for all intents and purposes, a layer N _is_ a new layer in one-at-a-time (and outside one-at-a-time that shouldn't happen in the first place).

So we'd have to change _all_ the scripts to take into account a _set_ of layers given a layer-index.

GregValiant, 2023-09-22:
@rburema - This happens whenever "One at a Time" is enabled.
I wrote a script that simply adds a delimiting line between the data sections.  It's "Add Data Headers" within the LittleUtilities post processor attached here.  I couldn't go back to a previous layer to pick up the last "E" value and I wondered why.

;TIME_ELAPSED:2584.373680
;>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>Start of DATA[204]<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
G1 F2100 E3686.33847
;>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>Start of DATA[205]<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
M107
G0 F9000 X48.194 Y44.012 Z24.998
G0 X159.685 Y52.07
;LAYER_COUNT:100
;>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>Start of DATA[206]<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
;LAYER:0
G0 X155.848 Y77.149 Z0.2

Depending on the retraction settings that single line may or may not be there.  When it is there, it is always a single line item in the Data List.
The travel move from the end of one model to the start of the next model is always present and is always it's own item in the Data List.  I tried merging those items into the previous layer item (Data[203] in this case) but Cura seemed to balk at getting a data list back that was shorter than the one sent out???
[LittleUtilities.zip](https://github.com/Ultimaker/Cura/files/12701690/LittleUtilities.zip)

GregValiant, 2026-10-01:
This seems to be a low priority so I'll close it.

closed 2026-10-01 by GregValiant