Ultimaker/Cura #15510: Vertical lines when using Fuzzy Skin on large cylinder with curved wall in 5.3.1

Dixitrix, 2023-05-12:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

When sliced, in Cura 5.3.1, a cylinder with curved walls over, a certain size (+- 100mm diameter), start to show horizontal lines when adding fuzzy skin as can be seen in the images. These horizontal lines might appear slight but they are very visible in printed parts. 

When slicing the same body without fuzzy skin there are no artefacts on the print. 
I have recreated the object, a simple revolve of a curve in Fusion 360 and every time I slice large bodies in 5.3.1 the vertical lines will appear. 

I have also done a fresh install of Cura 5.3.1 and the problem is still there. 

I have checked all setting from 5.3.1 and 5.2.2 and there is no difference other then the new setting that was not present in 5.2.2

I have sliced the same object in 5.2.2 and the vertical line are not present. 
 
[image]
[image]

### Actual results

Vertical lines on Fuzzy Skin

### Expected results

I have sliced the same object in 5.2.2 and the vertical line are not present. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Fuzzy curve body test for error 90mm Diameter.zip](https://github.com/Ultimaker/Cura/files/11467976/Fuzzy.curve.body.test.for.error.90mm.Diameter.zip)
[Fuzzy curve body test for error 180mm Diameter.zip](https://github.com/Ultimaker/Cura/files/11467977/Fuzzy.curve.body.test.for.error.180mm.Diameter.zip)

GregValiant, 2023-05-12:
Thanks for the report.
I'll leave this for the Cura Team.  There has been talk of removing Fuzzy Skin from Cura and if that is true then this probably won't be fixed.

MariMakes, 2023-05-22:
Hey @GregValiant,

Fuzzy Skin is one of my favorite features of Cura. If it's removed I'll be sad that it's gone but our GitHub would flood with requests to bring them back. We can see in the slicing data that it's used often. Luckily, it's not going anywhere. 👍 

@Dixitrix, I showed your issue to the team. They pointed out it's related to the problem we've been our troubleshooting with microsegments. We are collecting the test cases in this ticket: https://github.com/Ultimaker/Cura/issues/14811

Cura is still creating random fuzziness, but on very tiny lines that seem to be generated in a pattern. I took a look and in Cura 5.2 or earlier you don't see the alignment/patterns. If you need to get unstuck you can download a previous version here: 
 https://github.com/Ultimaker/Cura/releases
 
[image]

I'll be closing this issue as a duplicate but you can follow the progress here:  https://github.com/Ultimaker/Cura/issues/14811

GregValiant, 2023-05-22:
@MariMakes Yea, that was my bad.  I believe it was "Wire Printing" that I read was going to be discontinued.

Vandresc, 2023-06-09:
So, unfortunately, the fix we applied for the micro segments doesn't address this issue. We created a new ticket on our backlog. Devs see CURA-10682 for our internal reference.