Ultimaker/Cura #15512: Slicing failure

leppy110, 2023-05-13:
### Application Version

5.1.0

### Platform

Mac OS 10.15.7

### Printer

Anycubic Vyper

### Reproduction steps

I imported an stl file. Then I multiplied it. Then I made changes to the settings for printing (type of supports, angle, etc.). Then I attempted to slice the file.

### Actual results

The file would not slice

### Expected results

The file should slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Please tell me how to send the file from WeTransfer. It needs an email address.

GregValiant, 2023-05-13:
Thanks for the report.
Github accepts zip files but has a limit of 25mb.  Use the "File | Save Project" command and then in Explorer right-click on the 3mf file and select "Send To | Compressed (zipped) Folder".  Post the zip folder here.
If the zip is larger than 25mb you will have to post in somewhere like Google drive and provide a download link.

A lot of reports that come in are actually problems with the model.  Check the model and make sure it is watertight.

leppy110, 2023-05-13:
It’s not the file. When I only had 1 of it on the build plate it sliced
just fine. However, when I put more than 1 copy of the file the slicing
failed. Your site mentioned using WeTransfer and it needs an email address
to send it to. If I put in this email address, will that work?

GregValiant, 2023-05-13:
@MariMakes should be in on Monday.  She should be able to help with that.

leppy110, 2023-05-15:
> Hey @leppy110 <https://github.com/leppy110>,
>
> Greg is right, I would love to see your project file.
> In WeTransfer you can also select Get Transfer Link.
> Can you share that link with us? Then we can take a look.
> [image: image]
> <https://user-images.githubusercontent.com/40423138/238300592-4f529062-574b-4652-bb7d-c2c99275ed53.png>

I accidentally deleted the file. I have been having so many issues with my
printer. If it happens again, I will definitely send you a link.

leppy110, 2023-05-15:
So, I tried the process again and again it failed to slice. Here is the
WeTransfer link. https://we.tl/t-NEdfsgTqxG

Hope this helps you fix things and maybe you can let me know how to get it
working.

Thanks,

Laura Rosario

GregValiant, 2023-05-15:
This is another head scratcher.  It sliced for me on the first try.  After this first attempt I changed some support settings and it still sliced.
With the 4 statues it's a 24hour print.  I think this has a better than average chance of failure (tall skinny support) so I'd go with one statue at a time.  That way you wouldn't have to print over-night either.
(Of course you have to get it to slice first!)
[image]

leppy110, 2023-05-15:
Do you know of a way to get the settings from a gcode? I tried finding it
online but the instructions I found, the options they said to use aren’t in
any of the menus of the version of Cura that I am using.

Thanks in advance for your help.

Laura Rosario

GregValiant, 2023-05-15:
There is a plugin in the Marketplace "Export HTML Cura Settings" that will export the settings.  I've never used it.
The Cura Keywords are listed [HERE](http://files.fieldofview.com/cura/Replacement_Patterns.html).  You could make entries in your StartUp or Ending Gcode line:
;Line Width = {wall_line_width}
;Initial Layer Hgt = {layer_height_0}
The keywords need to be in curly brackets.

leppy110, 2023-05-15:
So I’m knot asking how to change the gcode.  I’m wanting to be able to find
out what settings were used in the slicer to make said gcode. Do you know
how to do that?

GregValiant, 2023-05-16:
You have to export them somehow, either to that HTML file or by listing the settings you are interested in within the Start or Ending Gcode as I explained.
When the gcode is created then Cura will dump the settings values (that you listed) into the gcode file and you will have a record of them.

At the end of any Cura gcode file is a section where every line starts with ";SETTINGS_3:".  Unfortunately, all that gets listed in that section are the _changes_ you made to the active setting profile rather than _all_ the settings contained in your profile.  It's fairly useless.

leppy110, 2023-05-16:
So I tried the HTML file. It isn't for already made gcodes. It is for when
you are making a project to save your settings as an HTML file so you can
reference them later. And I don't feel comfortable messing with Start or
Ending code because the last time I did anything with any kind of
programming was when I was in 1st grade on my Commodore 64.